LibreDWG/libredwg #870: Define AutoCAD release versions in include/dwg.h

michal-josef-spacek, 2023-11-24:
We have a situation, when we have DWG_VERSION_TYPE definitions in include/dwg.h, but this is related to DWG file, not to AutoCAD release.

Intent behind this is, that we need to define AutoCAD version and set defaults related to AutoCAD version.
e.g.
We have AutoCAD R_13 and AutoCAD R_2000.  Boths could encode DWG file to AC1012 version.
But result is different e.g. in case of sections. This is common problem.
AutoCAD R_13:
```
num_sections: 3 [RL]
num_sections => 3
section[0].number:         0 [RC] AcDb:Header
section[0].address:       70 [RL]
section[0].size:         454 [RL]
section[1].number:         1 [RC] AcDb:Classes
section[1].address:      524 [RL]
section[1].size:          38 [RL]
section[2].number:         2 [RC] AcDb:Handles
section[2].address:     3977 [RL]
section[2].size:          79 [RL]
crc: A24C [RSx] from 0-52
         HEADER (end):      70
```

AutoCAD R_2000:
```
num_sections: 5 [RL]
num_sections => 5
section[0].number:         0 [RC] AcDb:Header
section[0].address:    18207 [RL]
section[0].size:         484 [RL]
section[1].number:         1 [RC] AcDb:Classes
section[1].address:    18691 [RL]
section[1].size:         475 [RL]
section[2].number:         2 [RC] AcDb:Handles
section[2].address:    25375 [RL]
section[2].size:         273 [RL]
section[3].number:         3 [RC] AcDb:ObjFreeSpace
section[3].address:    25648 [RL]
section[3].size:          53 [RL]
section[4].number:         4 [RC] AcDb:Template
section[4].address:    25847 [RL]
section[4].size:           4 [RL]
crc: DA60 [RSx] from 0-70
         HEADER (end):      88
```

michal-josef-spacek, 2023-11-24:
I have prepared part of commit: https://github.com/michal-josef-spacek/libredwg/commit/41e41a1d398ae0bcc5018d59013a5049d266f4f2

I am looking for blank.DWG files from some AutoCAD releases:
* [ ] AutoCAD Release 2006
* [ ] AutoCAD Release 2008
* [ ] AutoCAD Release 2009
* [ ] AutoCAD Release 2010
* [ ] AutoCAD Release 2011
* [ ] AutoCAD Release 2012
* [ ] AutoCAD Release 2013
* [ ] AutoCAD Release 2014
* [ ] AutoCAD Release 2015
* [ ] AutoCAD Release 2016
* [ ] AutoCAD Release 2017
* [ ] AutoCAD Release 2018
* [ ] AutoCAD Release 2019
* [ ] AutoCAD Release 2020
* [ ] AutoCAD Release 2021
* [ ] AutoCAD Release 2022
Or other, if exists.

When you could help me, please send DWG file and release from which was saved. (Open AutoCAD and save)

rurban, 2023-11-24:
I"m sorry, I don't have a single AutoCAD anymore :) It died with my mac

turtle0x1, 2023-11-24:
I can probably help with the last 4 versions if need be, just empty DWG files? 

Be happy to pop them over to ruban in email if it would help?

michal-josef-spacek, 2023-11-24:
@turtle0x1 I will be happy. Only save the blank DWG file, archive it (zip, rar, gzip), and push to this ticket as an attachment.

turtle0x1, 2023-11-25:
Have abit of concern posting them publicly, not sure on the legal implications? (These are work licenses, and I have to assume there is some identifier in the dwgs)

michal-josef-spacek, 2023-11-25:
@turtle0x1 You could send it to me (michal.josef.spacek@gmail.com)

btw: This is probably a question to @rurban . I don't know about any license identification in the DWG file.

rurban, 2023-11-25:
The license not, but username and ID yes

michal-josef-spacek, 2023-11-25:
> The license not, but username and ID yes

AcDb:Header:
* FINGERPRINTGUID: "{254D4028-7034-9B4F-923B-5FE8735C027C}" [TU 1]
* VERSIONGUID: "{FC10ADCE-D03D-5B4D-B6A7-B25A1C6025A6}" [TU 1]
AcDb:SummaryInfo:
* LASTSAVEDBY: "Joshua" [TU 1]

Do you mean this?

rurban, 2023-11-26:
Exactly, plus the Security section

rurban, 2023-11-27:
We already have AUTOCAD_VERSION as our R_ release enums. But the extra table would be good to match it to the numeric dwg_version and maint_versions.

michal-josef-spacek, 2023-11-27:
> We already have AUTOCAD_VERSION as our R_ release enums. But the extra table would be good to match it to the numeric dwg_version and maint_versions.

Yes, but this is not used as AUTOCAD_VERSION, but as DWG version.

timoria21, 2023-11-29:
> I am looking for blank.DWG files from some AutoCAD releases:

Would some files written by OpenDesign SDK help in this? I can write: 

AC1012 R13
AC1014 R14
AC1015 R2000
AC1018 R2004
AC1021 R2007
AC1024 R2010
AC1027 R2013
AC1032 R2018

michal-josef-spacek, 2023-11-29:
> Would some files written by OpenDesign SDK help in this? I can write:

I am definitely interested in how different SW saves the DWG file.

timoria21, 2023-11-29:
Here you go Michal:

[blank_ODA.zip](https://github.com/LibreDWG/libredwg/files/13499604/blank_ODA.zip)

michal-josef-spacek, 2023-11-29:
@timoria21 Thank you very much.

rurban, 2023-11-30:
FYI: oda only discriminates on the dwg_version (only sometimes on the maint_version) and with class objects only on the class_version. Whilst we map the AC10xx magic to our own release numbers. But I have never seen a conflict so far

timoria21, 2023-11-30:
@michal-josef-spacek This is plain ODA SDK DWG output

rurban, 2023-11-30:
See, that's why we use the AC10xx magic, and not the dwg_version, which is just the version with which this file was created, and not with which version it was stored.

michal-josef-spacek, 2023-11-30:
Added AutoCAD 2018 and 2023, commit updated

michal-josef-spacek, 2023-11-30:
Added AutoCAD 2013, commit updated

michal-josef-spacek, 2023-12-04:
Added AutoCAD 2004, 2004 SP1a, and 2007, commit updated

rurban, 2024-02-11:
There is also the problem with the new VERSION checks, which could be a bit simplified to check the dwg_version, not the magic header version. ODA does only check this dwg_version.

I really want to have only R_2000 - R_2002 being 0x17, R-2004a - R_2004c being 0x18 and so on. They dd bump the dwg_version for each major dwg release, but it would be still only two versions to check.

michal-josef-spacek, 2024-02-11:
> That's your @michal-josef-spacek **add_autocad_version** branch, right?

Yes, that's mine..

michal-josef-spacek, 2024-02-11:
I am for merging and working on it.

rurban, 2024-02-11:
It doesn't pass the checks yet