LibreDWG/libredwg #875: dxfread?

fr-an-k, 2023-11-25:
I would like to convert dxf to/from json, but somehow there is no dxfread program.

I tried to make it by modifying dwgread to use dxf_read_file, which appears to be a drop in replacement for dwg_read_file since it appears to load to the same LibreDWG memory structure, but I get the invalid DWG error 0x800 on my DXF files that work fine with ezdxf.
The DXF version in case is AC1032 (> 2017).

Also, LibreDWG fails to decode its own DXF files generated from dwg (i.e. when running dwgwrite on them).

michal-josef-spacek, 2023-11-26:
When you need to convert DXF to JSON over LibreDWG you could use this:
```
dwgwrite -I DXF FILE.dxf
dwgread -O JSON -o FILE.json FILE.dwg
```

First command will create DWG file and second JSON file.

**NOTE1**: dwgwrite has support for AutoCAD 2000 now.
**NOTE2**: And there are issues still. We are resolving Second Header issues in AutoCAD 2000 now.

fr-an-k, 2023-11-26:
Thanks, but I get the same error without an explanation why, it's unable to decode the DXF (AC1032 = version 2018) while other programs and ezdxf have no issue with it:

<details>

```
Warning: Unknown HEADER.TEXTSTYLE ARIAL dxf:7
Warning: Unknown HEADER.CLAYER mk-nw-vlakken dxf:8
Warning: Unknown HEADER.DIMSTYLE M1000T25-PIJL dxf:3
Warning: Unknown HEADER.DIMTXSTY NLCS-ISO dxf:7
Warning: Unknown HEADER.CMLSTYLE Standard dxf:2
Warning: Unknown eed[].handle for APPID.ACAD_NAV_VCDISPLAY
Warning: Unknown eed[].handle for APPID.AcAecLayerStandard
Warning: Unknown eed[].handle for APPID.ACAD_DSTYLE_DIMJAG
Warning: Unknown eed[].handle for APPID.ACAD_DSTYLE_DIMTALN
Warning: Unknown LAYER.ltype Continuous
Warning: Unknown LAYER.ltype XX-HIDDEN
Warning: Unknown LAYER.ltype Continuous
ERROR: Failed to decode DXF file: testfiles/dxf/dxf/210.19.584-260688-S-2-xxxx-Duikers-191007.dxf

Warning: No class for GEODATA type 1080 found
ERROR 0x800
```

</details>

With -v9 option:
[out.txt](https://github.com/LibreDWG/libredwg/files/13467746/out.txt)

The file in question (zipped):
[210.19.584-260688-S-2-xxxx-Duikers-191007.zip](https://github.com/LibreDWG/libredwg/files/13467740/210.19.584-260688-S-2-xxxx-Duikers-191007.zip)

fr-an-k, 2023-11-26:
This error is because ```CHK_geomesh_pts``` fails (without logging why), apparently the number of points given upfront is less than what is actually there.

If I disable the check it ends with:
```malloc(): corrupted top size```
Disabling the check probably caused the excess geomesh pts to be written to unallocated memory.

Guess I'll have to fix the file somehow, before passing it to dwgwrite.

michal-josef-spacek, 2023-11-26:
I think, that there is a bug:
```
  dxf (93, 112)
GEODATA.num_geomesh_pts = 112 [BL 93]
```
Spec:
```
13,23 Coordinate of source mesh point (repeat)
14,24 Coordinate of destination mesh point (repeat)
```
but
```
  dxf (13, -6227056.729426)
  dxf (23, -5919056.729426)
GEODATA.geomesh_pts[0] = (-6227056.729426, -5919056.729426) [2RD 13]
  dxf (14, -36.964455)
  dxf (24, -8.675254)
GEODATA.geomesh_pts[1].dest_pt = (-36.964455, -8.675254) [2RD 13]
```
Two indexes from 112 and on the end is out of struct.
```
  dxf (14, -9.324418)
  dxf (24, 23.373242)
GEODATA.geomesh_pts[111].dest_pt = (-9.324418, 23.373242) [2RD 13]
  dxf (13, 1742539.799285)
ERROR: Failed to decode DXF file: 210.19.584-260688-S-2-xxxx-Duikers-191007.dxf
```

fr-an-k, 2023-11-26:
Yes I just submitted a pr #877

fr-an-k, 2023-11-26:
I will submit a pr for dxfread as well, so people can create a json from dxf directly.
(Although I'm not yet able to convert the json back to dxf succesfully.)

rurban, 2023-11-27:
north_dir also had a wrong type 3BD, but it is a 2RD really

rurban, 2023-11-27:
We don't want a seperate dxfread program just to export to JSON. See Michals workaround.

> Also, LibreDWG fails to decode its own DXF files generated from dwg (i.e. when running dwgwrite on them).

Yes, some more known bugs still.

closed 2023-11-27 by rurban