LibreDWG/libredwg #876: npm/WASM module, i.e. javascript bindings

fr-an-k, 2023-11-26:
Would be nice if libreDWG can be make available through npm or if instructions for compiling to WASM (without stdio dependency) were added.

That would make it available to a multitude of developers. WASM probably has a broader reach.

(Currently I'm stuck on incompatible glibc in the target environment, so this complicates everything, maybe the above wouldn't actually solve that)

fr-an-k, 2023-11-26:
Compiling with emscripten should yield a wasm file and js file, and it should use node fs for stdio.

However, I get stuck:

```
sh autogen.sh
emconfigure ./configure
```

output:
<details>

```
configure: ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking git version... d73eebde
checking so version-info... 0:13:0
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... /home/frank/repos/emsdk/upstream/emscripten/emcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /home/frank/repos/emsdk/upstream/emscripten/emcc accepts -g... yes
checking for /home/frank/repos/emsdk/upstream/emscripten/emcc option to enable C11 features... none needed
checking whether /home/frank/repos/emsdk/upstream/emscripten/emcc understands -c and -o together... yes
checking dependency style of /home/frank/repos/emsdk/upstream/emscripten/emcc... gcc3
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by /home/frank/repos/emsdk/upstream/emscripten/emcc... /home/frank/repos/emsdk/upstream/emscripten/emcc
checking if the linker (/home/frank/repos/emsdk/upstream/emscripten/emcc) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /home/frank/repos/emsdk/upstream/bin/llvm-nm
checking the name lister (/home/frank/repos/emsdk/upstream/bin/llvm-nm) interface... BSD nm
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /home/frank/repos/emsdk/upstream/emscripten/emcc option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... /home/frank/repos/emsdk/upstream/emscripten/emranlib
checking command to parse /home/frank/repos/emsdk/upstream/bin/llvm-nm output from /home/frank/repos/emsdk/upstream/emscripten/emcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
[... 161 lines cut ...]
checking whether the linker accepts -Wl,-z,relro... yes
checking whether the linker accepts -Wl,-z,now... yes
checking whether the linker accepts -Wl,-z,now... yes
checking whether the linker accepts -Wl,-z,noexecstack... yes
checking whether the linker accepts -Wl,-z,noexecstack... yes
checking whether the linker accepts -Wl,--no-as-needed... no
checking whether the linker accepts -Wl,--fatal-warnings... yes
checking whether the linker accepts -Wl,-fatal_warnings... no
checking for __attribute__((visibility("default")))... yes
checking whether C compiler accepts -Werror... yes
checking for __attribute__((gnu_format))... no
checking for __attribute__((ms_format))... no
checking for __attribute__((format))... no
checking for __attribute__((malloc))... no
checking for __attribute__((returns_nonnull))... no
checking for __attribute__((noreturn))... no
checking for __attribute__((aligned))... no
checking for printf() modifier to use with size_t... unknown
configure: error: cannot find a suitable modifier
emconfigure: error: './configure' failed (returned 1)
```

</details>

I looked at the related location in the configure script but I don't know what to make of that. Any ideas?

rurban, 2023-11-27:
You need to disable the m4/ax_printf_size_t.m4 probe on enscripten

closed 2023-11-29 by rurban

qwuip789, 2025-01-03:
configure: WARNING: unrecognized options: --enable-libconvert-only