LibreDWG/libredwg #891: Issue with handles in preR13

michal-josef-spacek, 2023-12-08:
Issue is from MC0.0.
Example (AC1.40):
[COMPLEX.DWG.gz](https://github.com/LibreDWG/libredwg/files/13620618/COMPLEX.DWG.gz)

from some entity
```
...
@offset 0x185
type: 14 [RS]
Add entity INSERT [21] Decode entity INSERT
===========================
Entity number: 21, Type: 14, Addr: 388
layer: 1 [H(RSd) 8]
object_map{2F} = 21
=> handle: (0.1.2F)
block_name: "BLOCK1" [TV 2]
ins_pt: (6, 1) [2RD 10]
scale: (0.5, 0.5) [2RD 41]
rotation: 0.523599 [RD 50]  30º

@offset 0x1b9
type: 10 [RS]
Add entity LOAD [22] Decode entity LOAD
===========================
Entity number: 22, Type: 10, Addr: 3bc
layer: 1 [H(RSd) 8]
object_map{2F} = 22
ERROR: Duplicate handle 2F for object 22 already points to object 21
=> handle: (0.0.2F)
file_name: "ES" [TV 1]

@offset 0x1c1
type: 4 [RS]
Add entity SHAPE [23] Decode entity SHAPE
===========================
Entity number: 23, Type: 4, Addr: 3c4
layer: 1 [H(RSd) 8]
object_map{2F} = 23
ERROR: Duplicate handle 2F for object 23 already points to object 21
=> handle: (0.0.2F)
ins_pt: (6, 6) [2RD 10]
scale: 1 [RD 40]
rotation: 0.523599 [RD 50]  30º
style_id: 131 [RS 0]
...
```

rurban, 2023-12-21:
Yes, but it's harmless. We only need the handles internally for table records, not any entities. And they are ignored on output <R13.

rurban, 2023-12-21:
Indeed not that harmless. dwg_next_handle was completely wrong, it scanned globals refs, but not the last objects for its handles. and esp. with preR13 we almost have no global refs.

closed 2023-12-23 by rurban