LibreDWG/libredwg #902: entity MTEXT member text reads only the first character

IdoGuzi, 2023-12-21:
I'm trying to extract the text from `MTEXT` entities from `.dwg` file.  
the text of the MTEXT in my file are 2 characters (example: 01, 02, etc),  
but I only get the first letter of the text.  

I'm on version 0.12.5 working on Ubuntu 22.04.
The version of dwg is AC1032.  

here is a snippet of how I'm trying to get the text: 
```
int main() {
    Dwg_Data dwg;
    dwg_read_file("my_file.dwg", &dwg);
    for (size_t i=0; i<dwg.num_objects; i++) {
        Dwg_Object object = dwg.object[i];
        switch (object.fixedtype) {
            case DWG_TYPE_MTEXT: {
                Dwg_Entity_MTEXT *mtext = object.tio.entity->tio.MTEXT;
		printf("text: %s\n", mtext->text);
            }
        }       
    }
}
```

I also tried looping over the pointer with size 2 (size of my text), but it will only print the first letter.

while I was trying to debug, I wanted to look at the raw data,
 but the address of the `Dwg_Object` gave the address: 275,739, 
although the size of the file is 114,537.
The comment of the address member in the header file says its the offset in the file,
but since its a to big of a number I'm guessing there is a problem with the way I'm handling things.

Hopefully it's not a bug and I'm just reading the text incorrectly, if it is a bug I'd by happy to help fix it.

rurban, 2023-12-22:
You are confused. >+ R2007 all text values are Windows unicode (i.e UCS-2),
With the rest you are also apparently confused. internal Dwg_Object * addresses are not offsets into the file.

closed 2023-12-22 by rurban

IdoGuzi, 2023-12-22:
Thanks for answering rurban.  
trying to get the text as 16 bit chars works now, I'm happy it's not a bug.  
Is there a place in the documentation taking about the text format?  
  
As I said in the post I'm on version 0.12.5 and I see the member's type has changed but the comment is the same.
quoting: "byte offset in the file".
https://github.com/LibreDWG/libredwg/blob/76d2f0d0fe699930ba0aa2f30beb921c0e81aea6/include/dwg.h#L8958

Can you please briefly explain to what this address is pointing.

rurban, 2023-12-22:
Dwg_Object *obj->address is indeed the byte offset. But you were talking about the obj ptr, not obj->address.

The docs at https://www.gnu.org/software/libredwg/manual/LibreDWG.html#Types mentions
 BITCODE_TU wchar*
    length + windows 2-byte wchar string (UCS-2). The default text type since r2007. 
    
    I..e. all T types are TU since r2007, else TV,

rurban, 2023-12-22:
`od --skip-bytes=275739 --read-bytes 192 my_file.dwg` would be easier.

IdoGuzi, 2023-12-22:
It's easier (thanks), but as I expected od gives:
`od: cannot skip past end of combined input`

iisaxin, 2023-12-25:
Try changing your dwg version to r2000,
Too high version to work properly

IdoGuzi, 2023-12-25:
> Try changing your dwg version to r2000,
> Too high version to work properly

I'm guessing you are talking about the object byte offset.  
The thing is it looks like the file is parsed correctly, so I don't think it's a version problem.  
After @rurban explained that the text isn't utf-8 (and it was a rookie mistake on my part) I'm extracting the text and all the other values just fine.  
The only problem I see is with the byte offset, though this problem doesn't affect my work, I would like to help make sure there's not a bug here.

iisaxin, 2023-12-25:
我的英语不太好，请尽量仔细的看懂我的话，
我的意思是你的dwg的文件版本太高，
你试试读取r2000版本的文件看看问题是否存在