LibreDWG/libredwg #908: skipped versions

fr-an-k, 2023-12-29:
It looks like (b) versions are skipped a lot, for instance in dwg_decode(), you have:

```
  PRE (R_13b1)
...
  VERSIONS (R_13b1, R_2000)
...
  VERSION (R_2004)
...
  VERSION (R_2007)
...
  SINCE (R_2010)
...

// This line should not be reached
...
```

meaning R_2000i, R_2002, R_2004a, R2004b, R2004c, R_2007a, R_2007b, R2010b get skipped and that last line gets reached when trying to decode one of those versions.

fr-an-k, 2023-12-31:
if this is indeed an issue, probably the easiest would be to fix it within the macro, for instance VERSION(R_2007) would check the set of the variants of that version, and SINCE would start at the first version variant.

michal-josef-spacek, 2024-02-09:
That's definitely issue.
I tested something this:
```
diff --git a/src/decode.c b/src/decode.c
index ded4bf2b6..d290cc139 100644
--- a/src/decode.c
+++ b/src/decode.c
@@ -226,15 +226,15 @@ dwg_decode (Bit_Chain *restrict dat, Dwg_Data *restrict dwg)
   {
     return decode_R13_R2000 (dat, dwg);
   }
-  VERSION (R_2004)
+  VERSIONS (R_2004a, R_2004)
   {
     return decode_R2004 (dat, dwg);
   }
-  VERSION (R_2007)
+  VERSIONS (R_2007a, R_2007)
   {
     return decode_R2007 (dat, dwg);
   }
-  SINCE (R_2010)
+  SINCE (R_2010b)
   {
     read_r2007_init (dwg); // sets loglevel only for now
     return decode_R2004 (dat, dwg);
```

closed 2024-02-11 by rurban