LibreDWG/libredwg #919: erro code 0x941

lym-ID, 2024-01-09:
geojson parsing, return error code "0x941", what is the meaning of this code?

michal-josef-spacek, 2024-01-09:
@lym-ID This kind of report isn't useful.

* There is no clean workflow, which you are using.
* We don't know the command, which you are using.
* We don't have the data on which you are testing.

If it is the error from libredwg, this is the composition of the following errors:
```
/**
 Error codes returned.
 */
typedef enum DWG_ERROR
{
  DWG_NOERR = 0,
  /* sorted by severity */
  DWG_ERR_WRONGCRC         = 1,
  DWG_ERR_NOTYETSUPPORTED  = 1 << 1, /* 2 */
  DWG_ERR_UNHANDLEDCLASS   = 1 << 2, /* 4 */
  DWG_ERR_INVALIDTYPE      = 1 << 3, /* 8 */
  DWG_ERR_INVALIDHANDLE    = 1 << 4, /* 16 */
  DWG_ERR_INVALIDEED       = 1 << 5, /* 32 */
  DWG_ERR_VALUEOUTOFBOUNDS = 1 << 6, /* 64 */
  /* -------- critical errors ------- */
  DWG_ERR_CLASSESNOTFOUND  = 1 << 7, /* 128 */
  DWG_ERR_SECTIONNOTFOUND  = 1 << 8, /* 256 */
  DWG_ERR_PAGENOTFOUND     = 1 << 9, /* 512 */
  DWG_ERR_INTERNALERROR    = 1 << 10,/* 1024 */
  DWG_ERR_INVALIDDWG       = 1 << 11,/* 2048 */
  DWG_ERR_IOERROR          = 1 << 12,/* 4096 */
  DWG_ERR_OUTOFMEM         = 1 << 13,/* 8192 */
} Dwg_Error;
```

Which is not very useful for user, because this means that something is very bad.

closed 2024-01-11 by lym-ID

michal-josef-spacek, 2024-01-11:
@lym-ID I understand, that document could be confidential.

Still:
* You could provide information about the process, which you are doing.
* We have e.g. ```dwgread -v3``` to create log record about document. Or part of log record, which contains error.
* When we could create simple dwg file with the same issue without confidential status, that's very helpful. This is not simple, but possible.

lym-ID, 2024-01-11:
This is the console log.
[image]

michal-josef-spacek, 2024-01-11:
@lym-ID Which version of libredwg do you use?

michal-josef-spacek, 2024-01-11:
@lym-ID And which version of DWG file is it?
First 6 bytes in the dwg file. Something like AC1015

lym-ID, 2024-01-11:
The last line has the version number

michal-josef-spacek, 2024-01-11:
> The last line has the version number

Ah, ok. Thanks

lym-ID, 2024-01-11:
dwg file version is AC1024

michal-josef-spacek, 2024-01-11:
@lym-ID I strongly recommend using a newer version of libredwg, e.g. https://github.com/LibreDWG/libredwg/releases/tag/untagged-da8a43d51bc7bdbd9d11
or development version.
There is strong development and this is the very old version.