LibreDWG/libredwg #924: Issue in MLINESTYLE object

michal-josef-spacek, 2024-01-24:
Example file:
[mlinestyle2.dwg.gz](https://github.com/LibreDWG/libredwg/files/14044142/mlinestyle2.dwg.gz)
[mlinestyle2.dxf.gz](https://github.com/LibreDWG/libredwg/files/14044143/mlinestyle2.dxf.gz)

Error:
```
Next object: 27 Handleoff: 0x2F [UMC] Offset: 235 [MC] @2779
==========================================
Object number: 27/1B, Size: 100 [MS], Type: 73 [BS], Address: 2781
Add object MLINESTYLE [27] Decode object MLINESTYLE
handle: 0.1.4C [H 5]
EED[0] size: 0 (end)
num_eed: 0
bitsize: 758 [RL]
 Hdlsize: 0x2A, hdl_dat: @94.6 - @100.0 (100)
num_reactors: 1 [BL 0]
ownerhandle: (4.1.E) abs:14 [H 330]
reactors[0]: (4.1.E) abs:14 [H* 330] => DICTIONARY
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 60
name: "FOO" [T 2]
description: "Description" [T 0]
flag: 592 [BS 70]
fill_color.index: 0 [CMC.BS 62]
start_angle: 1.5708 [BD 51]  90º
end_angle: 1.5708 [BD 52]  90º
num_lines: 6 [RC 71]
ERROR: Invalid MLINESTYLE.lines x 6

 object_map{4C} = 27
 padding: 80/0 (6 bits)
crc: BBCE [RSx]
 check_CRC 2779-2881 = 102: BBCE == BBCE
```

The error is because code in src/dec_macros.c:
```
...
#define _REPEAT(times, nam, type, idx)                                        \
  REPEAT_CHKCOUNT_LVAL (nam, _obj->times, type)                               \
  if (_obj->times > 0)
...
#define REPEAT_CHKCOUNT_LVAL(name, times, type)                               \
  if (AVAIL_BITS (dat) < 0)                                                   \
    {                                                                         \
      LOG_ERROR ("Invalid %s." #name ". No bytes left.\n", SAFEDXFNAME);      \
      times = 0;                                                              \
      if (_obj->name)                                                         \
        {                                                                     \
          free (_obj->name);                                                  \
          _obj->name = NULL;                                                  \
        }                                                                     \
      return DWG_ERR_VALUEOUTOFBOUNDS;                                        \
    }                                                                         \
  LOG_INSANE ("REPEAT_CHKCOUNT_LVAL %s." #name " x %lu: %lu > %" PRId64       \
              "?\n",                                                          \
              SAFEDXFNAME, (unsigned long)times,                              \
              (unsigned long)((times) * sizeof (type)), AVAIL_BITS (dat));    \
  if ((int64_t)((times) * sizeof (type)) > AVAIL_BITS (dat)                   \
      || (sizeof (times) > 4 && times > 0xc0000000 / sizeof (type)))          \
    {
...
```
The ```REPEAT_CHKCOUNT_LVAL``` not working.
in include/dwg.h is:
```
/**
 MLINESTYLE (73) object
 */
typedef struct _dwg_MLINESTYLE_line
{ 
  struct _dwg_object_MLINESTYLE *parent;
  BITCODE_BD  offset;
  BITCODE_CMC color;
  BITCODE_BSd lt_index;   /* until 2018 */
  BITCODE_H   lt_ltype;   /* since 2018 */
} Dwg_MLINESTYLE_line;
```

In real e.g. R14 is used structure without ```BITCODE_H   lt_ltype;```
However, the check compares the size of the full structure to the data.

When I comment ```REPEAT_CHKCOUNT_LVAL (nam, _obj->times, type)```, log is without error.

rurban, 2024-02-04:
I've changed it to a union, which passes this example

closed 2024-02-04 by rurban

michal-josef-spacek, 2024-02-05:
Seems fixed, thanks :+1:

commit aef43205 (2024-02-04): spec: fix MLINESTYLE_line.lt_ fields to union

because of the tricky overflow check in REPEAT_CHKCOUNT_LVAL.
Fixes #924
commit 55da496d (2024-02-04): spec: fix MLINESTYLE_line.lt_ fields to union

because of the tricky overflow check in REPEAT_CHKCOUNT_LVAL.
Fixes #924

WIP add_test. dxf from/to 2018 needs a converter.
commit d9e961c2 (2024-02-04): spec: fix MLINESTYLE_line.lt_ fields to union

because of the tricky overflow check in REPEAT_CHKCOUNT_LVAL.
Fixes #924

WIP add_test. dxf from/to 2018 needs a converter.