LibreDWG/libredwg #931: building libredwg 0.13 error, `error: 'calloc' sizes specified with 'sizeof' in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]  3421 |     dwg->header.section = (Dwg_Section *)calloc (sizeof (Dwg_Section)`

topazus, 2024-02-04:
build log:
<details>

```bash
❯ ./configure --disable-static --disable-bindings --disable-python && make -j6
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking git version... 0.13
checking so version-info... 0:13:0
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
[... 300 lines cut ...]
  CC       common.lo
  CC       codepages.lo
  CC       bits.lo
  CC       decode.lo
  CC       decode_r11.lo
  CC       decode_r2007.lo
dwg.c: In function 'dwg_sections_init':
dwg.c:3421:58: error: 'calloc' sizes specified with 'sizeof' in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
 3421 |     dwg->header.section = (Dwg_Section *)calloc (sizeof (Dwg_Section),
      |                                                          ^~~~~~~~~~~
dwg.c:3421:58: note: earlier argument should specify number of elements, later size of each element
cc1: all warnings being treated as errors
make[4]: *** [Makefile:735: dwg.lo] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/tmp/libredwg-0.13/src'
make[3]: *** [Makefile:754: all-recursive] Error 1
make[3]: Leaving directory '/tmp/libredwg-0.13/src'
make[2]: *** [Makefile:596: all] Error 2
make[2]: Leaving directory '/tmp/libredwg-0.13/src'
make[1]: *** [Makefile:712: all-recursive] Error 1
make[1]: Leaving directory '/tmp/libredwg-0.13'
make: *** [Makefile:591: all] Error 2
```

</details>

rurban, 2024-02-04:
Which gcc?
You need to use --disable-werror for now
Thanks

topazus, 2024-02-04:
@rurban I am using GCC 14.

With the build from https://github.com/LibreDWG/libredwg/commit/d70095adc25ce88a887a37d9956f2df668bc2457 , it also failed (not use --disable-werror for now), see the following build log

<details>

```bash
❯ ./configure --disable-static --disable-bindings --disable-python && make -j20
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking git version... 0.13.0.7003
checking so version-info... 0:13:0
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
[... 1261 lines cut ...]
encode.c:7607:27: error: allocation of insufficient size '3' for type 'Dwg_Eed_Data' {aka 'struct _dwg_entity_eed_data'} with size '25' [-Werror=alloc-size]
 7607 |       oo->eed[idx].data = (Dwg_Eed_Data *)calloc (3, 1);
      |                           ^
encode.c:7614:27: error: allocation of insufficient size '9' for type 'Dwg_Eed_Data' {aka 'struct _dwg_entity_eed_data'} with size '25' [-Werror=alloc-size]
 7614 |       oo->eed[idx].data = (Dwg_Eed_Data *)calloc (9, 1);
      |                           ^
encode.c:7633:27: error: allocation of insufficient size '3' for type 'Dwg_Eed_Data' {aka 'struct _dwg_entity_eed_data'} with size '25' [-Werror=alloc-size]
 7633 |       oo->eed[idx].data = (Dwg_Eed_Data *)calloc (3, 1);
      |                           ^
encode.c:7640:27: error: allocation of insufficient size '3' for type 'Dwg_Eed_Data' {aka 'struct _dwg_entity_eed_data'} with size '25' [-Werror=alloc-size]
 7640 |       oo->eed[idx].data = (Dwg_Eed_Data *)calloc (3, 1);
      |                           ^
cc1: all warnings being treated as errors
make[4]: *** [Makefile:735: encode.lo] Error 1
make[4]: Leaving directory '/home/aurora/fedora-src/libredwg/src'
make[3]: *** [Makefile:754: all-recursive] Error 1
make[3]: Leaving directory '/home/aurora/fedora-src/libredwg/src'
make[2]: *** [Makefile:596: all] Error 2
make[2]: Leaving directory '/home/aurora/fedora-src/libredwg/src'
make[1]: *** [Makefile:712: all-recursive] Error 1
make[1]: Leaving directory '/home/aurora/fedora-src/libredwg'
make: *** [Makefile:591: all] Error 2
```

</details>

rurban, 2024-02-04:
Oh, normally it's the uneven gcc numbers which are broken. But this is 14.
 This requires --disable-werror now for sure

closed 2024-02-04 by rurban