LibreDWG/libredwg #934: config.status: error: cannot find input file: `bindings/Makefile.in' when `./configure`

topazus, 2024-02-05:
using the build source: https://github.com/LibreDWG/libredwg/releases/download/0.13/libredwg-0.13.tar.xz
see:
<details>

```bash
❯ ./configure 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking git version... 0.13
checking so version-info... 0:13:0
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether ln -s works... yes
checking whether make sets $(MAKE)... (cached) yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
[... 267 lines cut ...]
checking for geojsonhint... no
checking for gjv... no
checking for jing... no
checking for valgrind... valgrind
checking for Valgrind tool memcheck... yes
checking for Valgrind tool helgrind... yes
checking for Valgrind tool drd... yes
checking VALGRIND version... 3.22.0
checking for Valgrind suppressions... valgrind-linux.supp
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating src/codepages/Makefile
config.status: creating doc/Makefile
config.status: creating programs/Makefile
config.status: creating examples/Makefile
config.status: creating test/Makefile
config.status: creating test/unit-testing/Makefile
config.status: creating libredwg.pc
config.status: creating vcpkg.json
config.status: error: cannot find input file: `bindings/Makefile.in'
```

</details>

rurban, 2024-02-05:
sh autogen.sh
would fix this. even automake.

closed 2024-02-06 by rurban

topazus, 2024-02-07:
@rurban Thanks for your quick reply and fix. But the [0.13.0.7050 version release](https://github.com/LibreDWG/libredwg/releases/download/0.13.0.7050/libredwg-0.13.0.7050.tar.xz) has another error when ./configure:

config.status: error: cannot find input file: `test/xmlsuite/Makefile.in'