LibreDWG/libredwg #937: Compilation issue with 0.13.1 on Linux

michal-josef-spacek, 2024-02-09:
```./configure --disable-bindings --enable-write --enable-trace --enable-debug```

```
...
checking for C/C++ restrict keyword... __restrict__
checking for ld... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking for shared library run path origin... done
./configure: line 13766: gl_HOST_CPU_C_ABI_32BIT: command not found
checking how to run the C preprocessor... gcc -E
checking for ELF binary format... yes
checking for the common suffixes of directories in the library search path... ./configure: line 13999: test: !=: unary operator expected
./configure: line 14005: test: !=: unary operator expected
./configure: line 14030: test: =: unary operator expected
./configure: line 14034: test: =: unary operator expected
lib,lib,lib
checking for iconv... yes
...
config.status: creating vcpkg.json
config.status: error: cannot find input file: `test/xmlsuite/Makefile.in'
...
```

SEE  [configure.log.gz](https://github.com/LibreDWG/libredwg/files/14223298/configure.log.gz)

[config.log.gz](https://github.com/LibreDWG/libredwg/files/14223301/config.log.gz)

danfe, 2024-02-10:
Yeah, looks like generated autotools bits got garbled during the bootstrap, e.g. without `--disable-bindings` it dies with similar `config.status: error: cannot find input file: 'bindings/Makefile.in'` message.  This is on FreeBSD, but I'm not opening another issue as it's likely the very same problem.  It happens with both versions 0.13.1 and original 0.13.

rurban, 2024-02-10:
Github creates the release versions by itself, from the --enable-release configuration. Have to check what happened yesterday. Official release will be today

rurban, 2024-02-10:
Shit, the github release process is broken. 
from its src tar.xz:
config.status: error: cannot find input file: `test/xmlsuite/Makefile.in'

apologies

closed 2024-02-10 by rurban