LibreDWG/libredwg #939: Improve releasing (see release 0.13.2)

michal-josef-spacek, 2024-02-10:
I saved release 0.13.2 from github release page (https://github.com/LibreDWG/libredwg/archive/refs/tags/0.13.2.tar.gz)
Issues:
1) There is no configure script in tarball.
2) When I run autogen.sh, there are many warning about not present git repository.

michal-josef-spacek, 2024-02-10:
Another thing:
```
  CC       in_json.lo
in_json.c:55:10: fatal error: ../jsmn/jsmn.h: Adresář nebo soubor neexistuje
   55 | #include "../jsmn/jsmn.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
```

rurban, 2024-02-10:
1. Fatal. But I tested it by myself, interesting.
2. Needs to be silenced somehow
3. Uh, forgot jsmn

rurban, 2024-02-12:
Unfortunately a wont-fix: https://github.com/orgs/community/discussions/6003

closed 2024-02-14 by rurban