LibreDWG/libredwg #941: Compiling issues with 0.13.2 in Linux/Debian

michal-josef-spacek, 2024-02-13:
```
woman:~> cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
woman:~/22/libredwg-0.13.2> ./configure --disable-bindings --enable-write --enable-trace --enable-debug --prefix=/usr/local/stow/libredwg-0.13.2
...
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking for shared library run path origin... done
./configure: line 13766: gl_HOST_CPU_C_ABI_32BIT: command not found
checking how to run the C preprocessor... gcc -E
checking for ELF binary format... yes
checking for the common suffixes of directories in the library search path... ./configure: line 13999: test: !=: unary operator expected
./configure: line 14005: test: !=: unary operator expected
./configure: line 14030: test: =: unary operator expected
./configure: line 14034: test: =: unary operator expected
lib,lib,lib
checking for iconv... yes
...
make
...
Making all in programs
make[2]: Entering directory '/home/skim/22/libredwg-0.13.2/programs'
make[2]: *** No rule to make target 'dwg2ps.1', needed by 'all-am'.  Stop.
make[2]: Leaving directory '/home/skim/22/libredwg-0.13.2/programs'
make[1]: *** [Makefile:710: all-recursive] Error 1
make[1]: Leaving directory '/home/skim/22/libredwg-0.13.2'
make: *** [Makefile:589: all] Error 2
```

[config.log.gz](https://github.com/LibreDWG/libredwg/files/14272702/config.log.gz)

rurban, 2024-02-14:
The GH release recipe misses libps also.

michal-josef-spacek, 2024-02-14:
This is not GH release (as I understand) = From http://ftp.gnu.org/gnu/libredwg/libredwg-0.13.2.tar.gz

rurban, 2024-02-14:
Fixed for the next release

rurban, 2024-02-14:
No, it is.
I always upload the GH generated releases to gnu. See HACKING/make release
Because we have no reproducible builds

danfe, 2024-02-26:
> Fixed for the next release

FWIW, I've reported exactly the same error with missing `dwg2ps.1` in version 0.12.5 (via email) and got the same reply.  I've been delaying FreeBSD port update for over two years now because of this, waiting for the fixed version.  Hope to finally see the manpage back in 0.13.3. ;-)

rurban, 2024-02-26:
Sorry, this wasn't checked.
You need to create your own dwg2ps.1 then. pod2man is your friend

closed 2024-02-26 by rurban