LibreDWG/libredwg #943: Issue in Windows with conversion between JSON and DWG

michal-josef-spacek, 2024-02-27:
The conversion from JSON to DWG is not same on Windows as on Linux.

Examples on libredwg.0.13.3:
[linux.tar.gz](https://github.com/LibreDWG/libredwg/files/14424125/linux.tar.gz)
[wine.tar.gz](https://github.com/LibreDWG/libredwg/files/14424126/wine.tar.gz)

Diff between dwgread output of Windows dwgrewrite and conversion versions:
```
1c1
< Reading DWG file line00-rewrite.dwg
---
> Reading DWG file out00.dwg
19,20c19,20
< section[2].address:    33136 [RL]
< section[2].size:         496 [RL]
---
> section[2].address:    32998 [RL]
> section[2].size:         498 [RL]
25c25
< section[4].address:    33780 [RL]
---
> section[4].address:    33644 [RL]
30c30
< crc: 1D68 [RSx] from 0-79
---
> crc: CA1D [RSx] from 0-79
272,273c272,273
< DIMLWD: -2 [BSd 70]
< DIMLWE: -2 [BSd 70]
---
> DIMLWD: -1 [BSd 70]
> DIMLWE: -1 [BSd 70]
316c316
< crc: 38FA [RSx] from 24955-25438=483
---
> crc: 68C1 [RSx] from 24955-25438=483
341,344c341,344
< =======> Handles (start) :    33136
<          Handles (end)   :    33632
<          Length: 496
< Handles page size: 490 [RS_BE]
---
> =======> Handles (start) :    32998
>          Handles (end)   :    33496
>          Length: 498
> Handles page size: 492 [RS_BE]
471c471
< Object number: 7/7, Size: 15 [MS], Type: 66 [BS]
---
> Object number: 7/7, Size: 18 [MS], Type: 66 [BS]
474c474
<  Hdlsize: 0x30,
---
>  Hdlsize: 0x48,
479c479
< num_entries: 2 [BS 70]
---
> num_entries: 3 [BS 70]
484,485c484,486
< crc: 5E17 [RSx]
<  check_CRC 17: 5E17 == 5E17
---
> entries[2]: (2.2.12D) abs:301 [H* 0]
> crc: A957 [RSx]
>  check_CRC 20: A957 == A957
801,802c802,803
< crc: 9910 [RSx]
<  check_CRC 65: 9910 == 9910
---
> crc: 7CBD [RSx]
[... 624 lines cut ...]
>  check_CRC 106: 0201 == 0201
> DEBUG_POS @33620.0 (268960)
> FFFFFF7F000000002B84DE31D76C6040ACDBBFF6EDC355FE
>   B  :1	  BB :3
>   RS :65535 / 0xffff (16)
>   BS :256 / 0x0100 (2)	
>   RL :2147483647 / 0x7fffffff (32)
>   RD :1.061e-314 (64)
> junk_r14: 0x7FFFFFFF [RLL 0]
>          Second Header (end)  : 33644
3835,3836c3444,3445
< =======> Template 4 (start)  :    33780
<          Template 4 (end)    :    33784
---
> =======> Template 4 (start)  :    33644
>          Template 4 (end)    :    33648
3841,3842c3450,3451
< num_objects: 89
< num_object_refs: 252
---
> num_objects: 90
> num_object_refs: 254
3844d3452
< Warning: Object handle not found 301/0x12D
```

michal-josef-spacek, 2024-02-27:
The first issue is the encoding of DIMLWD. Orginally -2, from wine JSON to DWG conversion -1
It's FIELD_BSd() macro.
In log:
right: 
```DIMLWD: 4294967294 [BSd]```
bad:
```DIMLWD: 2147483647 [BSd]```

michal-josef-spacek, 2024-02-27:
The second problem:
wine version adds extra APPID table
```
Next object: 89 Handleoff: 0x1 [UMC] Offset: 24 [MC] @32967
==========================================
Object number: 89/59, Size: 25 [MS], Type: 67 [BS], Address: 32969
Add table record APPID [89] Decode table record APPID
bitsize: 164 [RL] @3.2
 Hdlsize: 0x24, hdl_dat: @20.4 - @25.0 (25)
handle: 0.2.12D [H 5]
EED[0] size: 0 (end)
num_eed: 0
num_reactors: 0 [BL 0]
ownerhandle: (4.1.9) abs:9 [H 330]
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 60
name: "LibreDWG" [T 2]
is_xref_ref: 1 [B 0]
is_xref_resolved: 0 [BS 0]
is_xref_dep: 0 [B 0]
xref: (5.0.0) abs:0 [H 0]
unknown: 0x0 [RC 71]
 padding: +4
 object_map{12D} = 89
 padding: 0/0 (4 bits)
crc: F3C1 [RSx]
 check_CRC 32967-32994 = 27: F3C1 == F3C1
```

michal-josef-spacek, 2024-02-27:
The third issue:
Different Second Header junk_r14 value:
Right:
```
junk_r14: 0x3615F9E5B5CE9B52 [RLL 0]
```
Bad:
```
junk_r14: 0x7FFFFFFF [RLL 0]
```

timoria21, 2024-03-07:
I see but they both still incur in the "Recovery Mode" issue? Right?

michal-josef-spacek, 2024-03-07:
> I see but they both still incur in the "Recovery Mode" issue? Right?

yes

timoria21, 2024-03-08:
My feeling is that you are so close to resolve the "Recovery Mode" issue but maybe I'm wrong. I also offered a donation many times if it could help.

michal-josef-spacek, 2024-03-09:
The first and fourth issue is related to two things:
1) Bad conversion from DWG to JSON (-2 → 4294967294)
2) long in in_json.c on Windows is 32bit, so 4294967294 is not possible in this type.

michal-josef-spacek, 2024-03-09:
1) In PR: https://github.com/LibreDWG/libredwg/pull/952

michal-josef-spacek, 2024-03-09:
@rurban What about 2) ?
I created https://github.com/michal-josef-spacek/libredwg/commit/47c43c20c1f5519bb34b63eccfcbc11d6bc2b84c for fix.

rurban, 2024-03-09:
> The second problem: wine version adds extra APPID table

Not a problem, but on purpose for unhandled classes roundtrips

rurban, 2024-03-09:
> The first and fourth issue is related to two things:
> 
>     1. Bad conversion from DWG to JSON (-2 → 4294967294)
> 
>     2. long in in_json.c on Windows is 32bit, so 4294967294 is not possible in this type.

this must be uint32_t, not long

rurban, 2024-03-09:
> @rurban What about 2) ? I created [michal-josef-spacek@47c43c2](https://github.com/michal-josef-spacek/libredwg/commit/47c43c20c1f5519bb34b63eccfcbc11d6bc2b84c) for fix.
> 
> I think that is better to not use long type when is different on systems.

Sure, but we need strtol() which returns long, even on windows.

rurban, 2024-03-09:
Should be fixed now

michal-josef-spacek, 2024-03-09:
> > The second problem: wine version adds extra APPID table
> 
> Not a problem, but on purpose for unhandled classes roundtrips

I don't understand. On Linux not present, with mingw/wine present. :-)

michal-josef-spacek, 2024-03-09:
> > > The second problem: wine version adds extra APPID table
> > 
> > 
> > Not a problem, but on purpose for unhandled classes roundtrips
> 
> I don't understand. On Linux not present, with mingw/wine present. :-)

ok, I see :-D
```
if (fixup)
{
   ...
   new_appid = add_LibreDWG_APPID (dwg);
   ...
}
```

michal-josef-spacek, 2024-03-10:
I am closing this ticket.

Tested on last git branch on Linux and Windows with ```./configure --disable-bindings --enable-write --enable-trace --
enable-debug``` and working.

There are issues in compilation without ```--enable-debug```, see follow up: https://github.com/LibreDWG/libredwg/issues/954

closed 2024-03-10 by michal-josef-spacek