LibreDWG/libredwg #946: Use dwglayers.exe(0.13.3) to output the layer names of the dwg file. READ ERROR  *.dwg : 0x540

xiuweihan, 2024-03-01:
dwg file 1: 
[894-v12.zip](https://github.com/LibreDWG/libredwg/files/14456002/894-v12.zip)

dwg file 2: subsequent additions

output：READ ERROR *.dwg: 0x540

michal-josef-spacek, 2024-03-02:
@xiuweihan Thanks for examples. 

There are two different issues. When you use dwgread in both cases exits with error but different.
I am for split to two issues.

xiuweihan, 2024-03-05:
> @xiuweihan Thanks for examples.
> 
> There are two different issues. When you use dwgread in both cases exits with error but different. I am for split to two issues.

Hope it will be fixed in future versions. If I encounter related DWG drawings, I will submit them to this issue.

michal-josef-spacek, 2024-03-05:
@xiuweihan I moved second DWG file to another issue (https://github.com/LibreDWG/libredwg/issues/948)

michal-josef-spacek, 2024-03-05:
Issue is something bad with section[11]:
```
section_info[11] fields:
size:            4307852197888
num_sections:    14032
max_decomp_size: 1048576000 / 0x3e800000
unknown:         32768
compressed:      1003 (1=no, 2=yes)
type:            2049
encrypted:       27943936 (0=no, 1=yes, 2=unknown)
name:
fixedtype:       0

ERROR: read_R2004_section_info out of range
Warning: Failed to find section_info[11] with type 1
ERROR: Failed to read compressed Header section
```