LibreDWG/libredwg #948: Issue in AC1032 with decompression of sections

michal-josef-spacek, 2024-03-05:
Reported by @xiuweihan in https://github.com/LibreDWG/libredwg/issues/946

[5172_v2.zip](https://github.com/LibreDWG/libredwg/files/14493740/5172_v2.zip)

Error:
```
=== Data Section (Section Info 2911) @2198900 ===
section_type: 0x4163003b [RLx 0]
decomp_data_size: 47892 [RL 0]
comp_data_size: 31437 [RL 0]
compression_type: 2 [RL 0]
checksum: 0xbf6ece37 [RLx 0]
ERROR: Invalid opcode 0x6 in input stream at pos 35258854
Warning: Failed to find section_info[0] with type 1
ERROR: Failed to read compressed Header section
Found no section_info[0] with type 9
ERROR: Failed to read uncompressed SummaryInfo section
Warning: Failed to find section_info[0] with type 3
ERROR: Failed to read compressed Classes section
Warning: Failed to find section_info[0] with type 7
ERROR: Failed to read compressed AcDbObjects section
Warning: Failed to find section_info[0] with type 2
ERROR: Failed to read uncompressed AuxHeader section
Found no section_info[0] with type 10
ERROR: Failed to read uncompressed Preview section
Found no section_info[0] with type 11
ERROR: Failed to read uncompressed AppInfo section
Found no section_info[0] with type 12
AppInfoHistory section not found
Found no section_info[0] with type 13
FileDepList section not found
Found no section_info[0] with type 14
Security section not found
Found no section_info[0] with type 8
RevHistory section not found
Found no section_info[0] with type 6
ObjFreeSpace section not found
Found no section_info[0] with type 5
ERROR: Template section not found
```

Issue is with unhandled opcode ```Invalid opcode 0x6 in input stream at pos 35258854```

rurban, 2024-03-05:
I think it's rather the usual "large file, section corruption bug", with lots of deleted objects.
Warning: Invalid sections: 0 != numgaps: 0 + numsections: 2910

rurban, 2024-03-05:
> @rurban Is there a way how to compile and use ACadSharp on Linux?

yes, I've fixed things for linux upstream. 
you need a proper dotnet (e.g. ubuntu or fedora, not from microsoft).
but no idea yet how load a file, or how to debug into it

michal-josef-spacek, 2024-03-06:
> ```shell
> dotnet build
> ACadSharp.Examples/bin/Debug/net6.0/ACadSharp.Examples ../libredwg/ti/gh948/5172_v2.dwg

Could you share, how to build ACadSharp on Linux?

michal-josef-spacek, 2024-03-06:
> > ```shell
> > dotnet build
> > ACadSharp.Examples/bin/Debug/net6.0/ACadSharp.Examples ../libredwg/ti/gh948/5172_v2.dwg
> > ```
> 
> Could you share, how to build ACadSharp on Linux?

Ach.. dotnet supports this .sln files, ok.