LibreDWG/libredwg #949: How to set linetype to entity?

michal-josef-spacek, 2024-03-05:
I have an example:
[red_line.tar.gz](https://github.com/LibreDWG/libredwg/files/14498409/red_line.tar.gz)

Intention: Create a line, change color to red, create ltype, and use it in the created line.

I am curious, that in the DWG file isn't set ltype in the LINE entity.

What am I doing wrong?

rurban, 2024-03-06:
isbylayerlt is 1 by default. you need to unset it.
for r2000+ also need to set ltype_flags to 0b11 (3) that a handle exists.
```
 VERSIONS (R_13b1, R_14) //ODA bug
    {
      FIELD_B (isbylayerlt, 0);
#ifdef IS_DECODER
      if (FIELD_VALUE (isbylayerlt))
        FIELD_VALUE (ltype_flags) = FIELD_VALUE (isbylayerlt) ? 0 : 3;
#endif
    }
```
=>
Object number: 28, Size: 0 [MS], Type: 19 [BS], Address: 2392
Encode entity LINE
handle: 0.1.26 [H 5] @3.2
num_eed: 0
preview_exists: 0 [B 0] @3.5
bitsize: 0 @2399.5
entmode: 2 [BB 0] @7.7
num_reactors: 0 [BL 0] @8.1
isbylayerlt: 0 [B 0] @8.2
nolinks: 1 [B 0] @8.3
color.index: 1 [CMC.BS 62] @9.5
ltype_scale: 1 [BD 48] @9.7
invisible: 0 [BS 60] @10.1
start: (0.000000, 0.000000, 0.000000) [3BD 10] @10.7
end: (100.000000, 100.000000, 0.000000) [3BD 11] @27.5
thickness: 0 [BT 39] @27.7
extrusion: (0.000000, 0.000000, 1.000000) [BE 210] @28.5
HANDLE_STREAM @28.5
-bitsize calc from HANDLE_STREAM 229 @28.5 (2392)
xdicobjhandle: (3.0.0) abs:0 [H 360] @29.5
layer: (5.1.10) abs:16 [H 8] @31.5
ltype: (5.1.25) abs:37 [H 6] @33.5
-size: 34 [MS] @2390
-bitsize: 229 [RL] @2395.5
padding: +3 [*B]
write CRC 3F56 [RSx] from 2390-2426 = 36

now just the ltype.name needs to be valid. spaces are forbidden I think

Encode object LTYPE
handle: 0.1.16 [H 5]
num_eed: 0
bitsize: 0 [RL 0] @7.4
num_reactors: 0 [BL 0] @7.6
name: "CONTINUOUS" [T 2] @20.0
is_xref_ref: 1 [B 0] @20.1
is_xref_resolved: 0 [BS 0] @20.3
is_xref_dep: 0 [B 0] @20.4
xref: (5.0.0) abs:0 [H 0] @1.0
description: "Solid line" [T 3] @32.6
pattern_len: 0 [BD 0] @33.0
alignment: 0x41 [RC 72] @34.0
numdashes: 0x0 [RC 73] @35.0
strings_area: (null) [TF 256 0]
HANDLE_STREAM @291.0
-bitsize calc from HANDLE_STREAM 2328 @291.0 (1506)
ownerhandle: (12.1.11) abs:5 [H 330] @2.0
xdicobjhandle: (3.0.0) abs:0 [H 360] @3.0

vs

Encode object LTYPE
handle: 0.1.25 [H 5]
num_eed: 0
bitsize: 0 [RL 0] @7.4
num_reactors: 0 [BL 0] @7.6
name: "New LTYPE" [T 2] @19.0
is_xref_ref: 1 [B 0] @19.1
is_xref_resolved: 0 [BS 0] @19.3
is_xref_dep: 0 [B 0] @19.4
xref: (5.0.0) abs:0 [H 0] @1.0
description: "" [T 3] @19.6
pattern_len: 0 [BD 0] @20.0
alignment: 0x41 [RC 72] @21.0
numdashes: 0x0 [RC 73] @22.0
strings_area: (null) [TF 256 0]
HANDLE_STREAM @278.0
-bitsize calc from HANDLE_STREAM 2224 @278.0 (2106)
ownerhandle: (12.1.20) abs:5 [H 330] @2.0
xdicobjhandle: (3.0.0) abs:0 [H 360] @3.0

michal-josef-spacek, 2024-03-06:
Ah, thank you, I will try.

rurban, 2024-03-06:
I've adding now this api and some table name checks.

michal-josef-spacek, 2024-03-06:
I checked the names, and you are right, something "a-zA-Z0-9-_$" and chars with diacritics.
Thanks for code

rurban, 2024-03-11:
Simplified API added