LibreDWG/libredwg #954: Fix MATERIAL class

michal-josef-spacek, 2024-03-10:
When I compile libredwg without ```--enable-debug``` (tested on mingw cross compilation). The conversion of DWG with MATERIAL class to JSON and back is not working. Output DWG file is not possible to open in AutoCAD.

This is partially related to the implementation of MATERIAL class.

rurban, 2024-03-10:
MATERIAL.gentexture and several other MATERIAL features are not yet supported. These should be disabled without DEBUG

michal-josef-spacek, 2024-03-10:
If I understand correctly, we need to stable MATERIAL class and after this, we could remove hacks for roundtripping for MATERIAL (https://github.com/LibreDWG/libredwg/blob/master/src/encode.c#L3277). Am I right?

rurban, 2024-03-10:
We won't get procedural Materials stable anytime soon, but the rest would be great. So encode should warn and disable it. Just enable it with enable-debug

rurban, 2024-07-06:
renaming IMAGE,WIPEOUT.size to image_size fixed the debug json roundtrip

rurban, 2024-07-06:
I've re-enabled MATERIAL (and WIPEOUT) in the branch smoke/material. No oda errors with r2000 json roundtrips anymore. But still oda errors with r2004-2018

michal-josef-spacek, 2024-11-19:
This is working for me.

closed 2024-11-19 by michal-josef-spacek