LibreDWG/libredwg #962: Mis-positioned values in XYZ of Insert->ins_pt?

zefreestijl, 2024-03-16:
Hi, I'm new to this library and am trying to get the point from insert entities to process in other application
with code like this:

```
  dwg_ent_insert* ins1;

  auto cadPT1 = ins1->ins_pt;
  SUVector3D vect1 = { cadPT1.x , cadPT1.y , cadPT1.z  };
  SUPoint3D pt1 = { cadPT1.x  cadPT1.y , cadPT1.z  };
```

somehow the points' XYZ value is not the same as it is when I check it in the AutoCAD,

like the first Point mis-placed Y and Z value, though it's easy to correct by simply rotate via x-axis,
the issue seems to be irregular, like the second point all the three values are mis-placed,

[image]

I've checked other entity type like mtext and line, which don't have this problem?

did I mis-operate or is this really a problem, is so how can I fix it?

Thank you :]

michal-josef-spacek, 2024-03-16:
@zefreestijl It is good to present more information about your issue.
As I understand you read some DWG file (**Which version?**) and you need e.g. print insertion point. Sure?

You could use dwgread tool ```dwgread -v3 __DWG_FILE__``` to check if the information about the point is in place.

rurban, 2024-03-16:
Seems to be a reverse extrusion vector, not pointing to 0,0,1 but 0,0,-1.
Is this the case here?

zefreestijl, 2024-03-17:
> @zefreestijl It is good to present more information about your issue. As I understand you read some DWG file (**Which version?**) and you need e.g. print insertion point. Sure?
> 
> You could use dwgread tool `dwgread -v3 __DWG_FILE__` to check if the information about the point is in place.

Hi, the dwg version is 2007, yes I'm trying to fetch the dwg data the redrawing the entities in another app,
and I've followed your suggestion to inspect the output from dwgread.exe,
it does have the same problem as I used via library..

[image]

zefreestijl, 2024-03-17:
> Seems to be a reverse extrusion vector, not pointing to 0,0,1 but 0,0,-1. Is this the case here?

Hi, thanks for the reply, 
so the problem might be caused by the block which have been rotated and located on different axis?
but the rotate rotation attribute shows in AutoCAD and extracted via library is both 0,

or the caused is from the push system of an array, (sorry I'm not familiar with C/C++ language..)
I tried to build the sample execution projects with cmake,
but only the two main library built successfully,
all the executions project showed some source link error like LNK2028 and LNK2019,
still need to figure that out when I have spare time.. 
so if the vector problem needs to be check through the library project,
I might not be able to do that..

michal-josef-spacek, 2024-03-17:
Extrusion vectors are different in this two INSERT entities:
```
ins_pt: (8998.97, -150, 40597) [3BD 10]
scale_flag: 2 [BB 0]
scale.x: 100 [RD 41]
scale: (100, 100, 100) [DD 41]
rotation: 0 [BD 50]  0º
extrusion: (-1, 0, 0) [3BD 0]
```
vs
```
ins_pt: (80885.6, -150, -53999.4) [3BD 10]
scale_flag: 2 [BB 0]
scale.x: 100 [RD 41]
scale: (100, 100, 100) [DD 41]
rotation: 0 [BD 50]  0º
extrusion: (0, -1, 0) [3BD 0]
```

zefreestijl, 2024-03-17:
> Extrusion vectors are different in this two INSERT entities:
>
> ```
> ins_pt: (8998.97, -150, 40597) [3BD 10]
> scale_flag: 2 [BB 0]
> scale.x: 100 [RD 41]
> scale: (100, 100, 100) [DD 41]
> rotation: 0 [BD [...]

Hi, thank you, 
I didn't notice there's an attribute called extrusion that also affect the coordinates,
thank you and rurban for your help,
now I know how to fix it by checking them :]

closed 2024-03-17 by zefreestijl