LibreDWG/libredwg #973: hatch entity miss data

insertintosb, 2024-04-26:
when I use the lib to read  dwg file,  hatch entity miss some data... the dwg file version is  r2018 。
It becomes normal when the version is  r2004 ~ r2007;
The CAD soft I use is ZWCAD 2020 , OS is WIN10 .
hatch2007
[image]

hatch2018
[image]

[hatch.zip](https://github.com/LibreDWG/libredwg/files/15126485/hatch.zip)

insertintosb, 2024-04-26:
the lib version i use is  libredwg-0.13.3.7233-win64.

rurban, 2024-05-01:
The biggest difference I see is that the 2018 hatch is associative, the 2007 hatch not.
2007 has much more info, 2018 stops processing the paths. Need to cross-check

insertintosb, 2024-05-05:
I just test  some  CAD files of  different  version  with same hatch config (all  are associative) once more;
 The results are as follows:
 r2004 have the entire information.
 r2007 hatch's name miss, only the first letter appear.
 r2018 hatch's name miss, only the first letter appear , and the difine lines also miss.
 I find that  the hatch's  define lines  arise even though the version is r2018  if the path contains no splines  by change,  the hatch's name is as before.