LibreDWG/libredwg #974: Cannot open an autocad2022 version dwg file.

Nico3201975, 2024-04-28:
As the description of the title.

Trying to use **dwg_read_file** open a dwg file created by autocad 2022.

Somehow, I got error of DWG_ERR_VALUEOUTOFBOUNDS which is 64 in interger.

But it works fine with previous versions like 2007, 2010, 2012 etc..

rurban, 2024-04-28:
If you send it me, I can have look. If not, not

Nico3201975, 2024-04-29:
> If you send it me, I can have look. If not, not

Here is the file.

[test_2022.zip](https://github.com/LibreDWG/libredwg/files/15144496/test_2022.zip)

Nico3201975, 2024-04-29:
> If you send it me, I can have look. If not, not

After looking at the demo to load a dwg file, I realized that if the error is not critical, even it's 64 but the file will still be retrieved.

But when I tried to **add_fingerprint**  followed by the given examples, I found that dwg.header.version has to be set **R_2000** ? 
[image]

I would like to call dwg_add_TEXT to add some text, then back to Autocad2022 openning it, 

but somehow it cannot be open, because it looks like the file being converted to old vertion unless I click repair the file, that will delete what I have just added fingerprint, which will not be found when second times trying to read this file.

Is there a way to bypass this strict version stuff,  I would be much appriciated. Thanks!

rurban, 2024-04-30:
r2022 uses just the r2018 format, nothing new.
But I found a bug in read_2004_compressed_section for the uncompressed AppInfo section:
`|| es.fields.address + 32 + info->size > max_decomp_size`
is wrong and unneeded.

rurban, 2024-05-01:
The AppInfo bug is fixed. libredwg can read r2022 just fine.
For the write part, yes. we can only write up to r2000 yet, work in progress.
For the recovery prompt, yes, that's expected. What's going on with the missing add_TEXT I need to have a look

closed 2024-05-01 by rurban