LibreDWG/libredwg #975: _LibreDWG.so missing from python bindings

Katyos, 2024-05-10:
Using the precompiled windows binaries for version 13.3. When I import the LibreDWG.py module the first thing it does is look for the _LibreDWG.so file. However, this file is nowhere to be found in either the site-packages folder or the zip file as a whole.

Is there another file in the package that can be pointed to to do the same job? If not, would it be possible to ensure that the .so file is included in the python library please?

rurban, 2024-07-23:
A windows release does have no _LibreDWG.so file. It's called lib/python3.11/site-packages/_LibreDWG-0.dll there. I'll see where this wrong .so thing comes from.

Nobujine, 2024-07-23:
See this from SWIG documentation:
https://www.swig.org/Doc1.3/Python.html#Python_nn9

klmmlk, 2025-01-18:
The downloaded win64 version from releases reported an error during local testing,  
Environment: win11 + conda + python3.11
[image]

I compiled _LibreDWG.pyd locally, but it still throws an error during testing.
```
lin changcheng@Aliem MSYS /home/linchangcheng/pylibredwg/bindings/python
# python -c "import LibreDWG"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:/msys64/home/linchangcheng/pylibredwg/bindings/python/LibreDWG.py", line 12, in <module>
    import _LibreDWG
ImportError: DLL load failed while importing _LibreDWG: 找不到指定的模块。

lin changcheng@Aliem MSYS /home/linchangcheng/pylibredwg/bindings/python
# tree
.
├── LibreDWG.py
├── Makefile
├── Makefile.am
├── Makefile.in
├── _LibreDWG.la
├── _LibreDWG.pyd
├── _LibreDWG_la-swig_python.lo
├── _LibreDWG_la-swig_python.o
├── __pycache__
│   └── LibreDWG.cpython-312.pyc
├── dwg.h
├── dwg_api.h
└── swig_python.c
```