LibreDWG/libredwg #978: The necessary files are missing when installing

oyoyogg, 2024-05-28:
Hello, I need to use files like out_json.h and hash.h when installing and compiling, but the current installation method does not include these related files. Do you have a good solution for this? Thank you

rurban, 2024-05-28:
Where are out_json.h or hash.h needed? These are purely internal headers, and don't need to be installed.

They should be in the tar ball or git repos

oyoyogg, 2024-05-29:
After installing, I want to use methods similar to those in out_json.h in my own development tool(to convert some object of dwg to JSON), but the installation directory only has dwg.h or dwg_api.h, does that mean I need to implement those methods in my own code?Thank you

oyoyogg, 2024-05-29:
> Where are out_json.h or hash.h needed? These are purely internal headers, and don't need to be installed.
> 
> They should be in the tar ball or git repos
https://github.com/LibreDWG/libredwg/issues/978#issuecomment-2135520678

> Where are out_json.h or hash.h needed? These are purely internal headers, and don't need to be installed.
> 
> They should be in the tar ball or git repos

After installing, I want to use methods similar to those in out_json.h in my own development tool(to convert some object of dwg to JSON), but the installation directory only has dwg.h or dwg_api.h, does that mean I need to implement those methods in my own code?Thank you

rurban, 2024-05-29:
Then you need to use the source dist, not the package.

closed 2024-05-29 by rurban