LibreDWG/libredwg #984: add dwgwrite flag to help avoid recovery mode but with data-loss

rurban, 2024-06-14:
as wanted in #346

rurban, 2025-03-07:
Problematic (see add oda checks, json and DXF roundtrips):

CELLSTYLEMAP required by TABLESTYLE.
MATERIAL.
VX <r2000

fr-an-k, 2025-03-08:
CELLMAPSTYLE/MATERIAL/XRECORD don't appear in a newly created R_2000 DWG by LibreDWG or AutoCAD (and converted to JSON).
They do appear if you load LibreDWG's empty DWG into AutoCAD and save it again as R_2000 (and convert to JSON).

Unfortunately files exported by AutoCAD are full of all kinds of stuff, and I don't have direct access.

AutoCAD seems to also use these; perhaps some of them should be added by dwg_new_Document:
"ACAD_COLOR"
"ACAD_DETAILVIEWSTYLE"
"ACAD_LAST_SAVED_VERSION_INFO"
"ACAD_MATERIAL"
"ACAD_MLEADERSTYLE"
"ACAD_SCALELIST"
"ACAD_SECTIONVIEWSTYLE"
"ACAD_TABLESTYLE"
"ACAD_VISUALSTYLE"

I'll have to check out AcadSharp to see if they can avoid the warning.

fr-an-k, 2025-03-09:
ACadSharp has the same recovery warning (without finding errors) in the empty R_2000 dwg it generates.

However, an empty R_2018 produced by it does not have the warning.
And if you add a line entity it doesn't have the warning.

So it looks line I'll have to use ACadSharp in production for now, without JSON.
It also compiles to WASM alhough I didn't test that yet.

Perhaps LibreDWG's R_2018 export development can advance with the inspiration of ACadSharp, for better adoption in production environments and less issues like this one.

rurban, 2025-03-22:
He approach would be to write a seperate encoder, as in ACadSharp, which leaves out many ents and objects. In some cases they end up with 0 entities, and this file can easily loaded by AutoCAD. Because it is empty