LibreDWG/libredwg #988: C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3799,22): error C2065: “ref”: 未声明的标识符 [Desktop\libredwg-master\build2\libredwg.vcxproj]

lyw308, 2024-06-27:
C:\Users\Liu\Desktop\libredwg-master\src\common.h(1,1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3477,1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3799,22): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3799,23): error C2224: “.obj”的左侧必须具有结构/联合类型 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3800,22): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3800,23): error C2224: “.handleref”的左侧必须具有结构/联合类型 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3801,22): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3801,23): error C2224: “.absolute_ref”的左侧必须具有结构/联合类型 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3802,50): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3802,50): warning C4133: “函数”: 从“int *”到“Dwg_Object_Ref *restrict ”的类型不兼容 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3815,48): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3815,49): error C2224: “.absolute_ref”的左侧必须具有结构/联合类型 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3824,1): error C2065: “ref”: 未声明的标识符 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
C:\Users\Liu\Desktop\libredwg-master\src\decode.c(3824,1): error C2224: “.absolute_ref”的左侧必须具有结构/联合类型 [C:\Users\Liu\Desktop\libredwg-master\build2\libredwg.vcxproj]
[image]

rurban, 2024-06-29:
Apparently the native codepage 936 cannot be used with Visual Studio. Nothing we can do about that

lyw308, 2024-06-29:
> Apparently the native codepage 936 cannot be used with Visual Studio. Nothing we can do about that

ok thanks your reply

rurban, 2024-07-06:
msvc needs /utf-8 to be set. now added

closed 2024-07-06 by rurban