LibreDWG/libredwg #990: test dwg error,CAD opening error

oyoyogg, 2024-07-01:
hello ,I saved the DWG2000 file with dwgrewrite and the CAD does not recognize the opening error
`./dwgrewrite -v9 -o ./example_2000_rewrite.dwg ./example_2000.dwg`

oyoyogg, 2024-07-01:
Can you provide a verification tool for DWG files to check for structure errors? thank you

rurban, 2024-07-01:
dwgread.c is such a tool. ODAFileConverter is another.

oyoyogg, 2024-07-01:
[Uploading test_file.zip…]()
The test file opens normally with CAD, but it cannot be opened after converting with a dwgrewrite.
`nohup ./dwgrewrite -v3 -o ./stadium1_plan_level6_2000_cad_rw.dwg ./stadium1_plan_level6_2000_cad.dwg >./spal_rw1.log

**log:**
Next object: 44871 Handleoff: 0x1 [UMC]
==========================================
Object number: 44871/AF47, Size: 84 [MS], Type: 7 [BS]
Add entity INSERT [44871] Decode entity INSERT
bitsize: 561 [RL] @3.2
 Hdlsize: 0x6F,
handle: 0.2.C57F [H 5]

num_eed: 0
preview_exists: 0 [B 0]
entmode: 2 [BB 0]
num_reactors: 0 [BL 0]
nolinks: 0 [B 0]
color.index: 256 [CMC.BS 62]
ltype_scale: 10 [BD 48]
ltype_flags: 0 [BB 0]
plotstyle_flags: 0 [BB 0]
invisible: 0 [BS 60]
linewt: 0x1d [RC 370]
xdicobjhandle: (3.0.0) abs:0 [H 360]
prev_entity: (4.2.C57E) abs:50558 [H 0]
next_entity: (4.2.C580) abs:50560 [H 0]
layer: (5.2.35C4) abs:13764 [H 8]
ins_pt: (-205953, 272724, 0) [3BD 10]
scale_flag: 1 [BB 0]
scale: (1, 1, 1) [DD 41]
rotation: -2.45077e+58 [BD 50]  -1.40419e+60º
ERROR: bit_read_BD: unexpected 2-bit code: '11'
extrusion: (-2.25534e+241, 0, 0) [3BD 0]
has_attribs: 0 [B 0]
block_header: (5.2.4FA) abs:1274 [H 0]
crc: 6517 [RSx]
 check_CRC 86: 6517 == 6517

Next object: 44872 Handleoff: 0x1 [UMC]
==========================================
Object number: 44872/AF48, Size: 84 [MS], Type: 7 [BS]
Add entity INSERT [44872] Decode entity INSERT
bitsize: 561 [RL] @3.2
 Hdlsize: 0x6F,
handle: 0.2.C580 [H 5]

num_eed: 0
preview_exists: 0 [B 0]
entmode: 2 [BB 0]
num_reactors: 0 [BL 0]
nolinks: 0 [B 0]
color.index: 256 [CMC.BS 62]
ltype_scale: 10 [BD 48]
ltype_flags: 0 [BB 0]
plotstyle_flags: 0 [BB 0]
invisible: 0 [BS 60]
linewt: 0x1d [RC 370]
xdicobjhandle: (3.0.0) abs:0 [H 360]
prev_entity: (4.2.C57F) abs:50559 [H 0]
next_entity: (4.2.C581) abs:50561 [H 0]
layer: (5.2.35C4) abs:13764 [H 8]
ins_pt: (-211786, 271419, 0) [3BD 10]
scale_flag: 1 [BB 0]
scale: (1, 1, 1) [DD 41]
rotation: -2.45077e+58 [BD 50]  -1.40419e+60º
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: bit_read_BD: unexpected 2-bit code: '11'
extrusion: (-3.44137e+236, 0, 0) [3BD 0]
has_attribs: 1 [B 0]
block_header: (5.2.500) abs:1280 [H 0]
ERROR: bit_read_RC buffer overflow at 83.1 + 1 > 84
first_attrib: NULL 4 [H 0]
ERROR: bit_read_RC buffer overflow at 83.1 + 1 > 84
last_attrib: NULL 4 [H 0]
ERROR: bit_read_RC buffer overflow at 83.1 + 1 > 84
seqend: NULL 3 [H 0]
crc: 60C0 [RSx]
 check_CRC 86: 60C0 == 60C0
`

rurban, 2024-07-02:
INSERT encoding errors

oyoyogg, 2024-07-02:
> INSERT encoding errors

Thanks for the reply, yes, I also found out, is there a better way to deal with this at present?

oyoyogg, 2024-07-02:
I found out through the CAD coordinate positioning that there should be an error at dwg_read_file, ins_pt the corresponding block could not be found on the CAD

rurban, 2024-07-02:
The DD 41 could be wrong, because the next rotation is clearly wrong.

rurban, 2024-07-05:
With scale_flag 1 we must not write scale.x, it defaults to 1.0

closed 2024-07-05 by rurban