LibreDWG/libredwg #996: Getting rid of "Recovery Mode"

timoria21, 2024-07-19:
Regarding this issue: https://github.com/LibreDWG/libredwg/issues/346

Material is broken, right? What about hardcoding a blob of data that belongs to the correct material so we can write all DWG file formats without "Recovery Mode"?

Or better, can you point me to the line of code that adds the material? Is it safe to remove it alone?

Thanks.

rurban, 2024-07-19:
> Regarding the issue: #346

All material objects are already not written. I'm re-enabling it currently
to see what's the real problem.

timoria21, 2024-07-24:
From [here](https://github.com/LibreDWG/libredwg/issues/346#issuecomment-2134901386) I had understood that there was an issue with Material definition:

michal-josef-spacek, 2024-07-29:
I need to retest with the actual version.

rurban, 2024-07-30:
> I need to retest with the actual version.

There's still the ACAD_MATERIALS dictionary to be fixed. And LAYER.plotstyle seems wrong

michal-josef-spacek, 2024-07-30:
Actually blocked by https://github.com/LibreDWG/libredwg/issues/998

timoria21, 2024-10-31:
Isn't #998 resolved now?

Thank you!

michal-josef-spacek, 2024-11-01:
ok, need another test of the actual situation.

timoria21, 2024-11-04:
Thanks. If resolved, LibreDWG could write DWG files for AutoCAD with no "Recovery Mode" in all DWG file formats!

rurban, 2024-11-04:
David J. Harris ***@***.***> schrieb am Mo., 4. Nov. 2024,
15:33:

> Thanks. If resolved, LibreDWG could write DWG files for AutoCAD with no
> "Recovery Mode" in all DWG file formats!
>
Certainly not. There are still too many unknowns.

timoria21, 2024-11-05:
I'm optimistic!

timoria21, 2024-11-15:
Another issue related to this topic is that all entities added after AutoCAD 2000 cannot be written without "Recovery Mode" because the file format didn't support them yet:
* SubdMesh
* Table
* MLeader
* ArcDimension

michal-josef-spacek, 2024-11-15:
@timoria21 You could help us, when you will prepare simple dwg file with this "problematic" entity (with minimal content) and send it to this thread or the special issue which is related to the concrete entity (the best).

timoria21, 2024-11-19:
I meant that these entities were added after the only LibreDWG file format that works without "Recovery Mode", so I cannot test them...

michal-josef-spacek, 2024-11-19:
@timoria21 I am not sure if you understand "Recovery Mode" situation. I believe this is some major failure, maybe not in the entities. I touch issues in SecondHeader and I know that this was one root cause. And we don't understand SecondHeader still.

timoria21, 2024-11-19:
You told me it depends on Material you still need to understand in detail. I guess that Material is defined in SecondHeader. Yes, those entities cannot be written without "Recovery Mode" indirectly because of the unknown SecondHeader.

michal-josef-spacek, 2024-11-19:
@timoria21 
Sorry, I am talking badly. Last time there was a problem with SecondHeader. It's fixed for now. See example.
But, there are other major sections, in which each error could cause "Recovery Mode".
The issue with MATERIAL was with debug mode and the breaking of major structures.
**I need to look again, but each work on this is time-consuming :-/**

Step by step on AC1015 blank dwg file:
Example of second header in dwgread:
```
=======> Second Header (start): 22706
size: 114 [RL 0]
address: 22690 [BL 0]
version: "AC1015" [TFF 11 0]
4143313031350000000000
is_maint: 0x6 [RC 0]
zero_one_or_three: 0x1 [RC 0]
dwg_version: 0x617 [BSx 0]
=> header dwg_version: 0x17
=> header maint_version: 0x6
codepage: 28 [RS 0]
num_sections: 6 [BS 0]
sections[0].nr: 0 [RCd 0]
 Header
sections[0].address: 18879 [BL 0]
sections[0].size: 600 [BL 0]
sections[1].nr: 1 [RCd 0]
 Classes
sections[1].address: 19479 [BL 0]
sections[1].size: 190 [BL 0]
sections[2].nr: 2 [RCd 0]
 Handles
sections[2].address: 22525 [BL 0]
sections[2].size: 112 [BL 0]
sections[3].nr: 3 [RCd 0]
 ObjFreeSpace
sections[3].address: 22637 [BL 0]
sections[3].size: 53 [BL 0]
sections[4].nr: 4 [RCd 0]
 Template
sections[4].address: 0 [BL 0]
sections[4].size: 0 [BL 0]
sections[5].nr: 5 [RCd 0]
 AuxHeader
sections[5].address: 97 [BL 0]
sections[5].size: 123 [BL 0]
num_handles: 14 [BS 0]
handles[0].num_hdl: 1 [RCd 0]
handles[0].nr: 0 [RCd 0]
hdl[0]: 0x2c [RC 0]
[0] HANDSEED: 0.1.2C
handles[1].num_hdl: 1 [RCd 0]
handles[1].nr: 1 [RCd 0]
hdl[0]: 0x1 [RC 0]
[1] BLOCK_CONTROL_OBJECT: 0.1.1
handles[2].num_hdl: 1 [RCd 0]
handles[2].nr: 2 [RCd 0]
hdl[0]: 0x2 [RC 0]
[2] LAYER_CONTROL_OBJECT: 0.1.2
handles[3].num_hdl: 1 [RCd 0]
handles[3].nr: 3 [RCd 0]
hdl[0]: 0x3 [RC 0]
[3] STYLE_CONTROL_OBJECT: 0.1.3
handles[4].num_hdl: 1 [RCd 0]
handles[4].nr: 4 [RCd 0]
hdl[0]: 0x5 [RC 0]
[4] LTYPE_CONTROL_OBJECT: 0.1.5
handles[5].num_hdl: 1 [RCd 0]
handles[5].nr: 5 [RCd 0]
hdl[0]: 0x6 [RC 0]
[5] VIEW_CONTROL_OBJECT: 0.1.6
handles[6].num_hdl: 1 [RCd 0]
handles[6].nr: 6 [RCd 0]
hdl[0]: 0x7 [RC 0]
[6] UCS_CONTROL_OBJECT: 0.1.7
handles[7].num_hdl: 1 [RCd 0]
handles[7].nr: 7 [RCd 0]
hdl[0]: 0x8 [RC 0]
[7] VPORT_CONTROL_OBJECT: 0.1.8
handles[8].num_hdl: 1 [RCd 0]
handles[8].nr: 8 [RCd 0]
hdl[0]: 0x9 [RC 0]
[8] APPID_CONTROL_OBJECT: 0.1.9
handles[9].num_hdl: 1 [RCd 0]
handles[9].nr: 9 [RCd 0]
hdl[0]: 0xa [RC 0]
[9] DIMSTYLE_CONTROL_OBJECT: 0.1.A
handles[10].num_hdl: 1 [RCd 0]
handles[10].nr: 10 [RCd 0]
hdl[0]: 0xb [RC 0]
[10] VX_CONTROL_OBJECT: 0.1.B
handles[11].num_hdl: 1 [RCd 0]
handles[11].nr: 11 [RCd 0]
hdl[0]: 0xc [RC 0]
[11] DICTIONARY_NAMED_OBJECT: 0.1.C
handles[12].num_hdl: 1 [RCd 0]
handles[12].nr: 12 [RCd 0]
hdl[0]: 0x17 [RC 0]
[12] DICTIONARY_ACAD_MLINESTYLE: 0.1.17
handles[13].num_hdl: 1 [RCd 0]
handles[13].nr: 13 [RCd 0]
hdl[0]: 0xd [RC 0]
[13] DICTIONARY_ACAD_GROUP: 0.1.D
crc: ACEE [RSx]
 check_CRC 22706-22810 = 104: ACEE == ACEE
```
I believe that major things for "Recovery issue" are right values like "AC1015", size, version numbers, maybe all.
Seems ok.

Example of AuxHeader in same file:
```
=======> AuxHeader:         97
         AuxHeader (end):  220
aux_intro: { 0xff 0x77 0x1 } [*RC]
dwg_version: 0x17 [RSx 0]
maint_version: 0x6 [RLx 0]
numsaves: 1 [RL 0]
minus_1: -1 [RLd 0]
numsaves_1: 1 [RS 0]
numsaves_2: 0 [RS 0]
zero: 0 [RL 0]
dwg_version_1: 0x17 [RSx 0]
maint_version_1: 0x6 [RLx 0]
dwg_version_2: 0x17 [RSx 0]
maint_version_2: 0x6 [RLx 0]
unknown_6rs: { 10 17134 10 17134 0 1 } [*RS]
unknown_5rl: { 0 0 0 0 0 } [*RL]
TDCREATE: [2460090, 45365472] 2023-05-25 10:53:15 [TIMERLL 0]
TDUPDATE: [2460090, 45370948] 2023-05-25 10:53:20 [TIMERLL 0]
HANDSEED: 0x2c [RLx 0]
plot_stamp: 0 [RL 0]
zero_1: 0 [RS 0]
numsaves_3: 1 [RS 0]
zero_2: 0 [RL 0]
zero_3: 0 [RL 0]
zero_4: 0 [RL 0]
numsaves_4: 1 [RL 0]
zero_5: 0 [RL 0]
zero_6: 0 [RL 0]
```
The result is mainly unknown. These values are counters for changes in DWG file, probably versions of last AutoCAD which was used. It is working somehow actually. But conversion from DWG to JSON and back to DWG could cause some error because we don't know the right structure. Without an understanding of the right structure, there is still a risk of some failure.

michal-josef-spacek, 2024-11-19:
@timoria21 I did some tests now:
Tested files:
* entities-2d.dwg from libredwg repo (AC1015 version)
* entities-3d.dwg from libredwg repo (AC1015 version)
* Circle_Original.dwg

Testing procedures:
* Convert from DWG to JSON and back to DWG
* dwgrewrite

Testing environment:
* libredwg on Linux
* mingw version of libredwg on Linux

Results:
There is no "Recovery issue" in all.

Issues:
* In DWG-JSON-DWG test, there are differences between Linux and Windows version still

michal-josef-spacek, 2024-11-19:
@timoria21 Do you have some example which is failing?

michal-josef-spacek, 2024-11-19:
I closed https://github.com/LibreDWG/libredwg/issues/954 , tested and seems ok.

michal-josef-spacek, 2024-11-19:
Side effect: https://github.com/LibreDWG/libredwg/issues/1040

michal-josef-spacek, 2024-11-19:
Another side effect: https://github.com/LibreDWG/libredwg/issues/1042

michal-josef-spacek, 2024-11-19:
@timoria21 If you don't have an example file that has this issue, I will close this ticket. :-)

michal-josef-spacek, 2024-11-19:
> There are about 20 test files in our suite which fail json round trips with oda

This not mean, that it is related to this ticket :-)

timoria21, 2024-11-20:
Thanks for all these tests. To recap:

Is there any other DWG file format "Recovery Mode" free apart from Acad2000 as of today?

Acad2000 does not support the following entities:
* SubdMesh
* Table
* MLeader
* ArcDimension

michal-josef-spacek, 2024-11-20:
> Is there any other DWG file format "Recovery Mode" free apart from Acad2000 as of today?

I don't know.
 
> Acad2000 does not support the following entities:
> * SubdMesh
> * Table
> * MLeader
> * ArcDimension

Acad2000 doesn't support many other entities. It means that the entity could be in AC1015 DWG file, but some AutoCAD 2000 versions couldn't handle them (without some plugins or not handle in any way)

btw:
I have AC1015 DWG files with 57 entities (not the final count :-). 
I don't know what SubdMesh precisely is. I don't have any AC1015 DWG file with TABLE entity.
I have ARC_DIMENSION and MULTILEADER entities in AC1015 DWG files, I could test them.

I think the best way is to close this ticket. If there are some failing DWG files, you could create issue for it.

rurban, 2024-11-20:
> Thanks for all these tests. To recap:
> 
> Is there any other DWG file format "Recovery Mode" free apart from Acad2000 as of today?
> 
> Acad2000 does not support the following entities:
> 
>     * SubdMesh
>     * Table
>     * MLeader
>     * ArcDimension

Wrong. Since r2000 all possible objects and entities are supported, you just need to have the corresponding ARX file. Else acad stores them as PROXIES, just the graphics to display and plot them. .

timoria21, 2024-11-29:
Thanks for the clarifications. I hope you can resolve the "Recover Mode" soon for some upper DWG file formats, too.

closed 2024-11-29 by timoria21