LibreDWG/libredwg #999: An issue with LibreDWG created r2004 DXF file.

UgisL, 2024-07-31:
Hi,

I am trying to convert a r2018 DWG file into r2004 DXF file and then carry out some modifications on the latter. I use the following command line prompt (both files attached at the end as zip archive):
dwg2dxf.exe --as r2004 2018.dwg --file 2018.dxf

However, the produced dxf file is apparently problematic. The AutoDesk TrueView application can not open it, and also ezdxf does not read it (because it is missing name tag, see discussion in https://github.com/mozman/ezdxf/issues/1136). Is this supposed to work, or is this a bug?

I have also tried all other export formats (r2000, r2007, r2010, r2013) and got the same issue.

I am using LibreDWG v0.13.3, downloaded Windows binaries.

[2018.zip](https://github.com/user-attachments/files/16441827/2018.zip)

rurban, 2024-08-16:
Cannot reproduce. There are only 2 BLOCK entities, both with proper names. 
`./libtool --mode=execute programs/dwg2dxf --as r2004 -y -v3 -o 2018_gh999_r2004.dxf ../ti/gh999/2018.dwg`
`ODAFileConverter . . ACAD2004 DWG 0 1 2018_gh999_r2004.dxf`
works fine, without any ODA conversion errors.

Which BLOCK exactly?

UgisL, 2024-08-21:
Thanks a lot, you managed to find the issue before I could look at it again. Unfortunately I have moved on from this issue, but I tried a quick check with nightly windows binaries to convert again and load in Autodesk Trueview 2025. The generated DXF file managed to crash the application, so there might be some other issue ...

However, currently I do not have time to look more into this, but perhaps in a couple of weeks or roughly a month I might need to come back to this.

rurban, 2024-08-22:
Fixed with 1aa183fd828c0456166b9c8834bfeb5618ef6911

closed 2024-08-22 by rurban