LibreDWG/libredwg #1001: ./dwg.spec:12342:5: error: format specifies type 'unsigned short' but the argument has type 'BITCODE_BL' (aka 'unsigned int') [-Werror,-Wformat]

billydotzhang, 2024-08-16:
System: macOS Monterey
Chip: Apple M2

I followed the README instructions.

When running `sudo make` or`sudo make install`, this error occurs.

The following image shows the error I encountered during execution.

[image]

rurban, 2024-08-16:
Very good catch, clang! Thanks

billydotzhang, 2024-08-16:
I try agin. Another bug kept me from going.

Is it necessary to `make`

I just need to implement the `dwg2dxf` functionality

How do I run it

[image]

billydotzhang, 2024-08-16:
@rurban  Excuse me, give me a hand~ Thanks~

rurban, 2024-08-16:
To error on `VALUE_RS(0, 360)` is a bit too strict for my taste, but ok. I'll fix that also.

billydotzhang, 2024-08-19:
@rurban   Good morning. I am back.  XD
[image]

rurban, 2024-08-19:
> @rurban <https://github.com/rurban> Good morning. I am back. XD
> image.png (view on web)

Now that's silly. What if you change the 0 to 0U?

billydotzhang, 2024-08-19:
XD，I annotated it.

but.... 
[image]

closed 2024-08-19 by billydotzhang