LibreDWG/libredwg #1002: Segmentation fault in dwg2dxf

billydotzhang, 2024-08-19:
Hello, I was testing some dwg files and found segmentation fault in dwg2dxf.

macOS Monterey 12.5 , M2

compile with

./autogen.sh && ./configure --disable-bindings && make && sudo make install

```
➜  libredwg git:(master) ✗ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in src
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in codepages
make[4]: Nothing to be done for `all'.
Making all in programs
make[2]: Nothing to be done for `all'.
Making all in examples
make[2]: Nothing to be done for `all'.
Making all in test
Making all in unit-testing
make[3]: Nothing to be done for `all'.
make[3]: Nothing to be done for `all-am'.
Making all in doc
make[2]: Nothing to be done for `all'.
make[2]: Nothing to be done for `all-am'.
```

file:

[1.dwg.zip](https://github.com/user-attachments/files/16659750/1.dwg.zip)

**fault message**

Warning: Object handle not found 118670592/0x712C500 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074184/0x709AB48 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074185/0x709AB49 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074186/0x709AB4A in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074567/0x709ACC7 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074555/0x709ACBB in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074568/0x709ACC8 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074564/0x709ACC4 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074565/0x709ACC5 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118074566/0x709ACC6 in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118562828/0x711200C in 21373 objects of max 0x712F8BC handles
Warning: Object handle not found 118575040/0x7114FC0 in 21373 objects of max 0x712F8BC handles
Writing DXF file /Users/zhangcheng/Documents/CAD/libredwg/dwg2dxf/dxf/1.dxf
ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0
[... 121 lines cut ...]
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic

billydotzhang, 2024-08-19:
**fault message 2**

Warning: Object handle not found 97/0x61 in 98946 objects of max 0xE25A67 handles
Warning: Object handle not found 98/0x62 in 98946 objects of max 0xE25A67 handles
Warning: Object handle not found 99/0x63 in 98946 objects of max 0xE25A67 handles
Warning: Object handle not found 100/0x64 in 98946 objects of max 0xE25A67 handles
Warning: Object handle not found 101/0x65 in 98946 objects of max 0xE25A67 handles
Writing DXF file /Users/zhangcheng/Documents/CAD/libredwg/dwg2dxf/dxf/D-EL-01.dxf
ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

Warning: Empty REGION.history_id

Warning: empty revision_guid
Warning: Empty REGION.history_id

Warning: empty revision_guid
Warning: Unstable Class object 508 MLEADERSTYLE (0xfff) 51/D8
Warning: Unstable Class object 504 TABLESTYLE (0xfff) 75578/D93192
Warning: Unstable Class object 505 MATERIAL (0x481) 75581/D96DBA
Warning: Unstable Class object 505 MATERIAL (0x481) 75632/D99168
Warning: Unstable Class object 505 MATERIAL (0x481) 75640/D99170
Warning: Unstable Class entity 552 DBCOLOR (0x1) 75824/DAE725
Warning: Unstable Class entity 552 DBCOLOR (0x1) 75825/DAE726
Warning: Unstable Class entity 552 DBCOLOR (0x1) 75826/DB0658
Warning: Unstable Class entity 552 DBCOLOR (0x1) 75827/DB0659
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 79801/E124E4
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 79803/E124E6
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
[... 47 lines cut ...]
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98882/E25816
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98886/E2581A
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98890/E2581E
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98894/E25822
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98898/E25826
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98902/E2582A
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98906/E2582E
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98910/E25832
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98914/E25836
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98918/E2583A
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98922/E2583E
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 0 ACDB_BLKREFOBJECTCONTEXTDATA_CLASS (0x0) 98926/E25842
Warning: Unknown object, skipping eed/reactors/xdic

billydotzhang, 2024-08-20:
@rurban  We look forward to your early reply！ 🙏🏻

rurban, 2024-08-21:
> @rurban We look forward to your early reply！ 🙏🏻

Sorry, pretty busy right now. Will have to wait

rurban, 2024-08-22:
Not repro with 27118c40aaa8f208c07b070c95b5d182d050850a

closed 2026-04-07 by rurban