LibreDWG/libredwg #1008: Using readdwg command in dwgadd causes an error.

khlukekim, 2024-09-05:
Hello, I am testing dwgadd program and got stuck at the beginning.

I tried to use 'readdwg' command to read an existing dwg file and add a line to it. 
So the commands look like this:

    readdwg "test.dwg"
    line (0 0 0) (1 1 0)

Then I got an error - `ERROR: readdwg seen, but DWG already exists`. 

From the source code, I think it checks if `dwg` is present

    static int
    dwg_add_dat (Dwg_Data **dwgp, Bit_Chain *dat)
    {
      ...
      dwg = *dwgp;
      ...
      if (memBEGINc (p, "readdwg"))
      {
        if (dwg)
        {
           LOG_ERROR ("readdwg seen, but DWG already exists");
           exit (1);
        }

And it looks like that `dwg` is always not null since it's `memset` before calling `dwg_add_dat()`.

    int
    main (int argc, char *argv[])
    {
      ...
      memset (&dwg, 0, sizeof (dwg));
      ...
      dwgp = &dwg;
      LOG_INFO ("dwgadd process %s\n", argv[i])
      if ((retval = dwg_add_dat (&dwgp, &dat)) == 0)

I think checking the first line of `dwg_add_dat` and run `readdwg` (or dxf, json) command should be done before memsetting `dwg`, if I'm not missing anything. 
I built it from the latest master branch ([27118c4]).

Since I'm not even sure that it is an intended behavior, a bug or my mistake, please share your opinion on this. 
Any help would be greatly appreciated.

rurban, 2024-09-05:
Yes, this looks an easy, to fix bug.
We ever tested read dwg since. We need a test case also.

rurban, 2024-09-13:
Thanks. This got broken with the refactoring for 0.13, and we never had testcases for the read* commands. Now fixed and added.

rurban, 2024-09-17:
Merged

closed 2024-09-17 by rurban