LibreDWG/libredwg #1019: Incomplete data in AcDbSubDMesh

timoria21, 2024-09-25:
When attempting to read the attached DWG file and convert it to DXF or back to DWG, the resulting files appear to be invalid. Upon examining the JSON output, I noticed that the AcDbSubDMesh entities contain incomplete data with missing faces and vertices.
This issue was encountered while using the latest version 0.13.3.7345. Could you please investigate this issue?

[Modern Elegant Chair and Table.zip](https://github.com/user-attachments/files/17127371/Modern.Elegant.Chair.and.Table.zip)

rurban, 2024-09-25:
wrong spec: 
```
dlevel: 0 [BS 71]
is_watertight: 0 [B 72]
num_subdiv_vertex: 3758161920 [BL 91]
ERROR: Invalid subdiv_vertex size 3758161920. Need min. 22548971520 bits for 3BD, have 3835517 for MESH.
num_vertex: 3221225472 [BL 92]
ERROR: Invalid vertex size 3221225472. Need min. 19327352832 bits for 3BD, have 3835483 for MESH.
num_faces: 65536 [BL 93]
```

rurban, 2024-09-27:
Actually the MESH spec was correct. bit_read_BLL for the preview_size was wrong.
Should be fixed with smoke/gh1019-bll ab5cadcaea5e4fd01d6c8a56f6664d4670d1c873

closed 2024-09-28 by rurban