LibreDWG/libredwg #1021: Error while creating an DXF and importing it to AutoCAD

Mannshoch, 2024-09-26:
Log of the converting is attached:
[log.log](https://github.com/user-attachments/files/17147042/log.log)

AutoCAD output:
```
DXF-ReadError
Error in in APPID Table
DXF-ReadError auf Line 106298.
Invalid or incomplete DXF input -- drawing canceled.
```

Some lines starting on Line: 106298
```
AT_JNT_�வ�வ齏跽풴ۂ㺘㻦칀ۂ￾￿
 70
     0
  0
APPID
  5
2AE7
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
AT_JNT_肘౐肘౐ᠺಜ풴ۥ䇴ሸ칀ۥ￾￿
 70
     0
  0
APPID
  5
2AE8
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
AUDIT_I_190703115940-0
 70
     0
  0
APPID
  5
2AE9
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
AT_JNT_⸰ె⸰ెᖳ껀풴۬锼䃭칀۬￾￿
 70
     0
  0
APPID
  5
2AEA
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
AT_JNT_﵈䂾﵈䂾ᖳ껀풴۬㺠䂪칀۬￾￿
 70
     0
  0
APPID
  5
2AEB
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
AT_JNT_⾙츰۬ᖳ껀풴۬㵠䂪칀۬￾￿
 70
     0
  0
APPID
  5
2AEC
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
```

I'm not a Developer, I'm a CAD User which uses FreeSoftware at home. I'm not allowed to provide the DWG or DXF of the file. But I'm willing to follow advises If you need me doing things to find the errors.

Environment:
- Windows 10 / libredwg-0.13.3.7345-win64 (AutoCAD 2025 for test of DXF)
- File from AutoCAD 2019
- Has Proxy Objects from AutoPlant (3D)
- The DWG File itself contains several xref which them selves containing xref (in total 80) and has a Layout with viewport.
- Important for me at the end is everything 3D 
- the resulting DXF has 40MB

rurban, 2024-10-04:
Without the `XX1704-LAY-0900-P11628.dwg` we cannot help

Mannshoch, 2024-10-04:
Sorry, I'm not allowed to provide the DWG or DXF of the file. But I'm willing to follow advises If you need me doing things to find the errors.

michal-josef-spacek, 2024-10-09:
@Mannshoch The best way is create some simple DWG file which you could send to ticket. 

e.g. This is APPID section, so use this application and save.
I think that could be something with unicode string in DXF 2 section.

Example to test:
1) New DWG file
2) (regapp "ADESK_4153322344") # Fill name which you have in your situation
3) dxfout

Mannshoch, 2024-10-11:
[dev.zip](https://github.com/user-attachments/files/17342779/dev.zip)

michal-josef-spacek, 2024-10-11:
@Mannshoch Could you open these files in AutoCAD and save as DXF and save to us?

michal-josef-spacek, 2024-10-11:
Files (AC1032):
* Platte.dwg
** Issue with reading of sections
* Stahl.dwg
** Issue with reading of sections
* Teile.dwg
** ERRORs in ASSOCPERSSUBENTMANAGER, PERSUBENTMGR, REGION, _3DSOLID, ACSH_EXTRUSION_CLASS, BLOCK_HEADER
* test-trichter.dwg
** Issue with reading of sections
* xx17.dwg
** ERRORs in BLOCKSTRETCHACTION, BLOCK_HEADER, ASSOCPERSSUBENTMANAGER, PERSUBENTMGR, MTEXT,

michal-josef-spacek, 2024-10-11:
Issue with sections is described at https://github.com/LibreDWG/libredwg/issues/948

Mannshoch, 2024-10-14:
Seems the worst possible case I have.

michal-josef-spacek, 2024-10-14:
@Mannshoch The best way is to define, what is the biggest problem, define the problem, have an example, and fix it.
Repeat for all issues. It's doable.