LibreDWG/libredwg #1030: call examples for dwg_dynapi_common_utf8text?

zenglanmu, 2024-10-12:
I am trying to build a "high-level" python wrapper for this library.
Similar to issuue #991, I was stuck in extracting utf8 string from Dwg_Obj.
For example, I have a Dwg_Layer obj, and want access its name field as utf8 string.
In the c code side, I called dwg_dynapi_common_utf8text function like following:
```
int isnew = 0;
char* textp;
bool result;
result = dwg_dynapi_common_utf8text(layer_obj, "name", &textp, &isnew, null);
```
but I got this error
`
LOG_ERROR ("%s: Invalid common text field %s", __FUNCTION__, fieldname);
`

Also using uppercase fieldname "NAME" is not work either. Is I missed something important or get utf8 name field is not supported for layer type obj?

zenglanmu, 2024-10-12:
Besides, following is the typemap for wrapper those dynapi utf8 functions which added in dwg.i file , if anyone is interested, 
```
%include "typemaps.i"  // common type mappings
// for support function like dwg_dynapi_entity_utf8text, which have char **restrict textp, int *isnewp as argout

%include "typemaps.i"  // common type mappings
// for support function like dwg_dynapi_entity_utf8text, which have char **restrict textp, int *isnewp as argout
%typemap(in,numinputs=0) (char **restrict textp, int *) (char* tmp1, int tmp2) %{
  $1 = &tmp1;
  $2 = &tmp2;
%}

%typemap(argout) (char **restrict textp, int *) (PyObject* py_textp, PyObject* py_int) %{
  // assuming the func return variable generate by swig is called 'result'
  if(result && *$1){
    py_textp = PyUnicode_FromString(*$1);
  }else{
    py_textp = Py_None;
    Py_INCREF(Py_None);
  }
  py_int = PyInt_FromLong(*$2);
  // we dont care isnew result, only want get string result
  $result = SWIG_Python_AppendOutput($result,py_textp);
%}

%typemap(freearg) (char **restrict textp, int *) %{
  // only free if isnew
  if (*$1 && *$2){
    free (*$1);
  }              
%}
// end of support function like dwg_dynapi_entity_utf8text
```

Unlike issue #991, it seems the free memory problem had been overcomed by mapping textp and isnew together.

rurban, 2024-10-25:
the LAYER object had no common field "name". it is in the object section.