LibreDWG/libredwg #1031: Fix DIMASSOC object

michal-josef-spacek, 2024-10-12:
The actual DIMASSOC object is bad.
This is about fixing the object.

The first PR is https://github.com/LibreDWG/libredwg/pull/1029

TODO:
- [ ] Fix osnap_type = 0
- [ ] How to work with items? I think the actual code is bad. Number of items is from 1 to 5 and probably depends on associativity and osnap_type, but how?
- [ ] The last point probably isn't defined

michal-josef-spacek, 2024-10-13:
I updated PR
1) fixed
2) updated to working situation. Need to rewrite to something better. @rurban any idea?
3) fixed

michal-josef-spacek, 2024-10-13:
Not tested in DXF and JSON.

michal-josef-spacek, 2024-10-15:
@rurban This PR is a work in progress in case 2) I don't know how to implement it better. I think the best way is to check the class name and if the name is "AcDbOsnapPointRef" then continue.

btw: I was looking to >AC1018 and there are issues with 2)

michal-josef-spacek, 2024-10-15:
The actual code fails on dxf2dwg in file: 
[dim1.dxf.gz](https://github.com/user-attachments/files/17377010/dim1.dxf.gz)

michal-josef-spacek, 2024-10-15:
ad osnap_pt) It's 3d point, see example file:
[dim4.dwg.gz](https://github.com/user-attachments/files/17379336/dim4.dwg.gz)

Output from actual code:
```
Next object: 64 Handleoff: 0x1 [UMC] Offset: 3227 [MC] @11024
==========================================
Object number: 64/40, Size: 98 [MS], Type: 503 [BS], Address: 11026
Warning: Unstable Class object 503 DIMASSOC (0x0) 64/0
Add object DIMASSOC [64] Decode object DIMASSOC
bitsize: 655 [RL] @4.2
 Hdlsize: 0x81, hdl_dat: @81.7 - @98.0 (98)
handle: 0.2.10F [H 5]
EED[0] size: 0 (end)
num_eed: 0
num_reactors: 1 [BL 0]
ownerhandle: (12.1.A) abs:261 [H 330]
reactors[0]: (4.2.105) abs:261 [H* 330] => DICTIONARY
xdicobjhandle: (3.0.0) abs:0 [H 360]
--common_size: 68
unknown_bits [698 (68,587,0) 88 TF]: 40C0088A0B1A23127B9B730B82837B4B73A2932B30320281504A6DD220C95FC27F5202815111416344624F736E6170506F696E745265660640502A095BF3ADFE0BF5CFEA40502A60852100829820FBA0FAA0FA20FB20FA02

associativity: 0x3 [BLx 90]
trans_space_flag: 0 [B 70]
rotated_type: 0x0 [RC 71]
dimensionobj: (4.1.F7) abs:247 [H 330]
DIMASSOC_Ref.rcount1: 0
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[0].osnap_type: 0x6 [RC 72]
ref[0].num_xrefs: 1 [BL 0]
ref[rcount1].xrefs[vcount]: (4.1.F5) abs:245 [H 331]
ref[0].main_subent_type: 2 [BL 73]
ref[0].main_gsmarker: 0 [BL 91]
ref[0].num_xrefpaths: 0 [BL 0]
ref[0].osnap_dist: 0.55266 [BD 40]
ref[0].osnap_pt: (0, 0, 0) [3BD 10]
ref[0].num_intsectobj: 1 [BL 0]
ref[rcount1].intsectobj[vcount]: (4.1.F4) abs:244 [H 332]
ref[0].intersec_subent_type: 2 [BL 74]
ref[0].intersec_gsmarker: 0 [BL 92]
ref[0].num_intersec_xrefpaths: 0 [BL 0]
ref[0].has_lastpt_ref: 0 [B 75]
DIMASSOC_Ref.rcount1: 1
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[1].osnap_type: 0x6 [RC 72]
ref[1].num_xrefs: 1 [BL 0]
ref[rcount1].xrefs[vcount]: (4.1.F6) abs:246 [H 331]
ref[1].main_subent_type: 2 [BL 73]
ref[1].main_gsmarker: 0 [BL 91]
ref[1].num_xrefpaths: 0 [BL 0]
ref[1].osnap_dist: 0.362303 [BD 40]
ref[1].osnap_pt: (0, 0, 0) [3BD 10]
ref[1].num_intsectobj: 1 [BL 0]
ref[rcount1].intsectobj[vcount]: (4.1.F4) abs:244 [H 332]
ref[1].intersec_subent_type: 2 [BL 74]
ref[1].intersec_gsmarker: 0 [BL 92]
ref[1].num_intersec_xrefpaths: 0 [BL 0]
ref[1].has_lastpt_ref: 0 [B 75]
 padding: +1
 object_map{10F} = 64
 padding: E9/1 (1 bits)
crc: D5F1 [RSx]
 check_CRC 11024-11124 = 100: D5F1 == D5F1
```

michal-josef-spacek, 2024-10-15:
I think that ``2e+50`` in ``ref[0].osnap_pt: (0, 0, 2e+50) [3BD 10]`` is the filling.

rurban, 2024-12-16:
work/dimassoc-pr1029 works for me now.  49b02a1b
my 2d osnap pt commit was wrong. and your loop size changes.

michal-josef-spacek, 2024-12-16:
@rurban ok, this seems nice.

The only thing is missing one AcDbOsnapPointRef block in case of any has_lastpt_ref = 1.
If I understand it correctly.

michal-josef-spacek, 2024-12-16:
Example:
```
--common_size: 68
unknown_bits [938 (68,814,0) 118 TF]: 40C008920B1A23127B9B730B82837B4B73A2932B300052028150470DD2E0CD41AA7F5112416344624F736E6170506F696E74526566000840502A2E254911821A79CFEAA2482C688C49EE6DCC2
E0A0DED2DCE8A4CACC001480A05411C374B833506A9FD4804220D8304220D742206A42202342206AF03

associativity: 0x3 [BLx 90]
trans_space_flag: 0 [B 70]
rotated_type: 0x0 [RC 71]
dimensionobj: (4.2.20D7) abs:8407 [H 330]
DIMASSOC_Ref.rcount1: 0
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[0].osnap_type: 0xa [RC 72]
ref[0].num_xrefs: 1 [BL 0]
ref[rcount1].xrefs[vcount]: (4.2.206A) abs:8298 [H 331]
ref[0].main_subent_type: 2 [BL 73]
ref[0].main_gsmarker: 0 [BL 91]
ref[0].num_xrefpaths: 0 [BL 0]
ref[0].osnap_dist: 0.337915 [BD 40]
ref[0].osnap_pt: (0, 0, 0) [3BD 10]
ref[0].has_lastpt_ref: 0 [B 75]
DIMASSOC_Ref.rcount1: 1
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[1].osnap_type: 0x8 [RC 72]
ref[1].num_xrefs: 1 [BL 0]
ref[rcount1].xrefs[vcount]: (4.2.2023) abs:8227 [H 331]
ref[1].main_subent_type: 2 [BL 73]
ref[1].main_gsmarker: 0 [BL 91]
ref[1].num_xrefpaths: 0 [BL 0]
ref[1].osnap_dist: 0.731571 [BD 40]
ref[1].osnap_pt: (0, 0, 0) [3BD 10]
ref[1].has_lastpt_ref: 1 [B 75]
 handle stream: +259 @77.7 MISSING (@110.2  @124.2)
 padding: +30 MISSING
 object_map{20D9} = 2054
 padding: D0/10 (6 bits)
crc: 0954 [RSx]
 check_CRC 2636457-2636587 = 130: 0954 == 0954
```

Another example:
```
--common_size: 60
unknown_bits [922 (60,856,0) 116 TF]: 40C008920B1A23127B9B730B82837B4B73A2932B300042028151F3FFFFFFFFFFDE7F5512416344624F736E6170506F696E745265660000A0000000000252C500040000000040AFC088920B1A2
3127B9B730B82837B4B73A2932B30000A02815344D323F62870DB0A520107ECC107E90699069F03

associativity: 0x3 [BLx 90]
trans_space_flag: 0 [B 70]
rotated_type: 0x0 [RC 71]
dimensionobj: (4.1.FA) abs:250 [H 330]
DIMASSOC_Ref.rcount1: 0
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[0].osnap_type: 0x8 [RC 72]
ref[0].num_xrefs: 1 [BL 0]
ref[rcount1].xrefs[vcount]: (4.1.A6) abs:166 [H 331]
ref[0].main_subent_type: 2 [BL 73]
ref[0].main_gsmarker: 0 [BL 91]
ref[0].num_xrefpaths: 0 [BL 0]
ref[0].osnap_dist: 1 [BD 40]
ref[0].osnap_pt: (0, 0, 0) [3BD 10]
ref[0].has_lastpt_ref: 1 [B 75]
DIMASSOC_Ref.rcount1: 1
ref[rcount1].classname: "AcDbOsnapPointRef" [T 1]
ref[1].osnap_type: 0x0 [RC 72]
ref[1].num_xrefs: 0 [BL 0]
ref[1].osnap_dist: 0 [BD 40]
ref[1].osnap_pt: (4500, -4000, 0) [3BD 10]
ref[1].has_lastpt_ref: 0 [B 75]
 handle stream: +259 @82.1 MISSING (@114.4  @122.4)
 padding: +20 MISSING
 object_map{FC} = 114
 padding: 64/4 (4 bits)
crc: EF4C [RSx]
 check_CRC 136575-136702 = 127: EF4C == EF4C
```

rurban, 2024-12-17:
> @rurban ok, this seems nice.
> 
> The only thing is missing one AcDbOsnapPointRef block in case of any has_lastpt_ref = 1. If I understand it correctly.

Ah, this could be it, good

closed 2026-11-16 by michal-josef-spacek