LibreDWG/libredwg #1044: Warning: checksum: 0x9e200de2 (calculated) CRC mismatch 0xf0b4-0xf135

LiStopaD45, 2024-11-20:
Hello! I encountered a strange error. I converted a file dwg to geojson 2 days ago. Everything worked fine. But today I wanted to convert another file and got an error. I tried to convert old files, but I got an error. I did not change the file with the library. I tried to download it again and even a different version. But the error haunts me. I ask for help. I give the error listing below.

`Warning: checksum: 0x9e200de2 (calculated) CRC mismatch 0xf0b4-0xf135

Warning: Invalid sections: 0 != numgaps: 0 + numsections: 17
Warning: Unstable Class object 504 TABLESTYLE (0xfff) 34/0
Warning: TODO TABLESTYLE r2010+ missing fields
Warning: Unstable Class object 505 MATERIAL (0x481) 46/0
Warning: Unstable Class object 505 MATERIAL (0x481) 47/0
Warning: Unstable Class object 505 MATERIAL (0x481) 48/0
Warning: Unstable Class object 508 MLEADERSTYLE (0xfff) 69/0
Warning: Unstable Class object 508 MLEADERSTYLE (0xfff) 76/0
ERROR: Invalid zero_18 size 3. Need min. 48 bits, have 0 for .
ERROR: Some section size or address out of bounds
Warning: AcDs schdat segment yet unhandled
Warning: AcDs _data_ segment yet unhandled
Warning: AcDs prvsav segment yet unhandled
Warning: AcDs freesp segment yet unhandled
Warning: Object handle not found 679/0x2A7
Warning: TODO TABLESTYLE r2010+ missing fields
Warning: AcDs schdat segment yet unhandled
SUCCESS`

michal-josef-spacek, 2024-11-21:
@LiStopaD45 You could help us to investigate this. Probably related to some concrete entity/object/item which is failing.
We need to know: 
* Version of original DWG file (AC string at the beginning of the file)
* Ideally DWG file or DWG file with an example entity, which has the issue.
* You could run ```dwgread -v3 __DWG_FILE__``` and look to entity, which has ERROR: invalid zero_18, probably SecondHeader and send to us. (-v3 means more verbose output)

rurban, 2025-04-19:
TODO Take that CRC calc from ACadSharp