LibreDWG/libredwg #1045: Warning: checksum: 0x935dbb45 (calculated) CRC mismatch 0xaafa14-0xab29e5

rikan1, 2024-11-21:
Hello , i encountered a issue when converting a large (12MB) dwg file to json .

 'Warning: checksum: 0x935dbb45 (calculated) CRC mismatch 0xaafa14-0xab29e5\n' +
    '\n' +
    'ERROR: Skip section AcDb:Preview with max decompression size 0x7d20 > 0x1800\n' +
    'ERROR: Skip section  with size 89 > 1 * 0\n' +
    'ERROR: read_R2004_section_info out of range\n' +
    'Warning: Failed to find section_info[9] with type 1\n' +
    'ERROR: Failed to read compressed Header section\n' +
    'Warning: Failed to find section_info[9] with type 3\n' +
    'ERROR: Failed to read compressed Classes section\n' +
    'Warning: Failed to find section_info[9] with type 4\n' +
    'ERROR: Failed to read compressed Handles section\n' +
    'Warning: Failed to find section_info[9] with type 2\n' +
    'ERROR: Failed to read uncompressed AuxHeader section\n' +
    'ERROR: Invalid section AcDb:Preview count or max decompression size. Sections: 1, Max size: 0\n' +
    'ERROR: Failed to read uncompressed Preview section\n' +
    'ERROR: Template section not found\n' +
    '\n' +
    'ERROR: Invalid num_segidx\n' +
    'ERROR: Failed to decode file: /Users/somyaranjankar/Desktop/code/bbs-dwg-api/src/temp/1731333834741.dwg 0x940\n' +
    '\n' +
    'ERROR 0x940\n'

dwgread 0.13.3.7338 is used for conversion.

michal-josef-spacek, 2024-11-24:
@rikan1 Hi, we need more information. 
1) You could send us the AC string from the beginning of the file.
2) ```dwgread -v3 __DWG_FILE__``` could help more.

santanup, 2024-11-27:
> @rikan1 Hi, we need more information.
> 
> 1. You could send us the AC string from the beginning of the file.
> 2. `dwgread -v3 __DWG_FILE__` could help more.

@michal-josef-spacek 

we are getting this
'Reading DWG file /home/ubuntu/bbs-dwg-api/src/temp/1732731851358.dwg\n' +
     "This file's version code is: AC1027 (r2013)\n" +
     'is_maint: 0x8 [RC 0]\n' +
     'zero_one_or_three: 0x3 [RC 0]\n' +
     'thumbnail_address: 576 [RL 0]\n' +
     'dwg_version: 0x21 [RC 0]\n' +
     'maint_version: 0x4 [RC 0]\n' +
     'codepage: 30 [RS 0]\n' +
     'ANSI_1252\n' +
     'unknown_0: 0x0 [RC 0]\n' +
     'app_dwg_version: 0x0 [RC 0]\n' +
     'app_maint_version: 0x0 [RC 0]\n' +
     'security_type: 0 [RL 0]\n' +
     'rl_1c_address: 0 [RL 0]\n' +

santanup, 2024-11-28:
> ok, we know that is `AC1027`

How to fix it? @michal-josef-spacek

michal-josef-spacek, 2024-11-28:
@santanup I don't know what your problem is. 
1) The first message is:
```Hello , i encountered a issue when converting a large (12MB) dwg file to json .```
What is the result?

michal-josef-spacek, 2024-11-28:
4) Issue:
```
ERROR: Skip section with size 89 > 1 * 0
```
Probably related to something bad with reading of sections.
I don't know how much this is important. Probably not much in your case.

michal-josef-spacek, 2024-11-28:
What is the result? So, some section errors, and some cannot block functionality. Something is probably very bad. Without an example DWG file we cannot fix it.

santanup, 2024-11-28:
> What is the result? So, some section errors, and some cannot block functionality. Something is probably very bad. Without an example DWG file we cannot fix it.

@michal-josef-spacek 
These are the files -
- https://d2k638bldkf5wf.cloudfront.net/importDwg/1732784727769.dwg
- https://d2k638bldkf5wf.cloudfront.net/importDwg/1732784624493.dwg

michal-josef-spacek, 2024-11-28:
@santanup Thank you.
Analysis:

a) 1732784727769.dwg
Version:  AC1027
Created by: Teigha
(be careful, Teigha has issues itself)
I see errors in PROXY_OBJECTs, DICTIONARY

b) 1732784624493.dwg
Version: AC1032
Created by: AutoCAD
I see many errors in many entities, In AC1032 we need to fix many things.

I could export both to JSON by:
```dwgread -O JSON __INPUT_DWG__ | tee __OUTPUT_JSON__```

What is the root cause, which you are trying to resolve?

santanup, 2024-11-28:
Okay we will try, thanks

rikan1, 2024-11-28:
@michal-josef-spacek 
Can you please check this file .
[https://d2k638bldkf5wf.cloudfront.net/importDwg/1731333834741.dwg](url)

dwgread -O JSON __INPUT_DWG__. This command is failing

michal-josef-spacek, 2024-11-28:
@rikan1 Ok, this is the real issue with no possibility of reading this file.

rikan1, 2024-11-28:
@michal-josef-spacek How to fix this , do you have any suggestions ?

michal-josef-spacek, 2024-11-30:
@rikan1 Check section parsing. This is not easy to do for me.

rurban, 2025-03-04:
With the latest large file 1732784727769.dwg I see this internal problem:

entities[13065]: (3.4.1A1D7E78) abs:438140536 [H* 0]
REALLOC dwg->object_ref vector to 32768
entities[13066]: (3.4.1A1D7E79) abs:438140537 [H* 0]Warning: Object handle not found 606492518/0x24265766

after realloc the refs, the handles cannot be found anymore.
...

rurban, 2025-03-04:
The first warning checksum: 0x935dbb45 (calculated) CRC mismatch 0xaafa14-0xab29e5 is in a lot of DWG's. looks the 2nd crc algo is still wrong, but this only affects writing 2004 formats, which we cannot do yet.

ignore it.

rurban, 2025-04-19:
The section info bugs are now fixed with #439 (the new decompressor)
The remaining ATTDEF with MTEXT and FIELD bugs are known limitations and will be fixed later. 
Also the object_ref realloc bug.