LibreDWG/libredwg #1051: These files won't open.

dotoritos-kim, 2024-11-30:
[drawings.zip](https://github.com/user-attachments/files/17966865/drawings.zip)
[A1-041~051 형별성능관계내역 - 업로드 불가.zip](https://github.com/user-attachments/files/17966870/A1-041.051.-.zip)

Attached files are included in the text. Both debina and windows do not open equally.

```bash
Reading DWG file /app/uploads/1732998991bILZwIkspIcH/A1-041~051 형별성능관계내역 - 업로드 불가.dwg
ERROR: Invalid System Section Page Map type 0x943c4008 != 0x41630e3b
ERROR: Failed to decode file: /app/uploads/1732998991bILZwIkspIcH/A1-041~051 형별성능관계내역 - 업로드 불가.dwg 0x100

READ ERROR 0x100
Reading DWG file /app/uploads/1732998991bILZwIkspIcH/A1-041~051 형별성능관계내역 - 업로드 불가.dwg.dxf
ERROR: File not found: /app/uploads/1732998991bILZwIkspIcH/A1-041~051 형별성능관계내역 - 업로드 불가.dwg.dxf

READ ERROR 0x1000
```

Since the drawings are used for actual work, can they be hidden once enough data has been shared or the problem has been resolved? I have the authority, but I don't want it to be made blatantly public.

rurban, 2025-03-04:
A1, A3*: Invalid System Section Page Map offsets.

=== Read System Section (Section Page Map) @ab98e0 ===

section_type: 0x943c4008 [RLx 0]
ERROR: Invalid System Section Page Map type 0x943c4008 != 0x41630e3b
ERROR: Failed to decode file: ../ti/gh1051/A1-041~051 형별성능관계내역 - 업로드 불가.dwg 0x100

A2 stops reading at 
< Next object: 27887 Handleoff: 0x1 [UMC] Offset: 47 [MC] @1575573
==========================================
Object number: 27887/6CEF, Size: 42 [MS]

Mannshoch, 2025-04-08:
I also wanted to read a file with special characters using dwgread. I also receive an ERROR 0x1000
I could only avoid the problem by changing the file name.

rurban, 2025-06-02:
All files but A2-051~055 펌프,전기,발전기실, 빗물저류조 평단면도 have an Invalid System Section Page Map type.
Cannot do much about that. Not a bug.

A2-051~055 펌프,전기,발전기실, 빗물저류조 평단면도 reads as SUCCESS, but has a lot of unknown AEC and AECC objects

closed 2025-06-02 by rurban