LibreDWG/libredwg #1053: Issue with BLOCKSTRETCHACTION object handles

michal-josef-spacek, 2024-12-02:
After fixing of https://github.com/LibreDWG/libredwg/issues/1049 there is another issue in BLOCKSTRETCHACTION
Example situation in dwg:
```
--common_size: 29
unknown_bits [1161 (29,1057,430) 146 TF]: 3FFFFFFFD1D41CF1D84251D41E25A7632EB9D0D9100AA483C5AEC2985C0903405014090240BD31232F40BCA110319634E4381E21100A2393B0B94A5F

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 7 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 9 [BL 0]
name: "Stretch" [TU 300]
be_major: 29 [BL 98]
be_minor: 7 [BL 99]
eed1071: 0 [BL 1071]
display_location: (162.122, -677.24, 0) [3BD 0]
num_deps: 3 [BL 71]
deps[0]: (4.2.4AD) abs:1197 [H* 330] => BLOCKLINEARGRIP End Grip
deps[1]: (4.2.4AB) abs:1195 [H* 330] => BLOCKBASEPOINTPARAMETER Base Point
deps[2]: (4.2.4A5) abs:1189 [H* 330] => LWPOLYLINE
num_actions: 1 [BL 70]
actions[0]: 1 [BL 91]
conn_pts[0].code: 2 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 2 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (670.251, 655.11) [2RD 1011]
pts[vcount]: (498.619, -144.805) [2RD 1011]
num_hdls: 1 [BL 73]
hdls[rcount1].hdl: (4.2.4A5) abs:1189 [H 331]
hdls[0].bs74: 2 [BS 74]
hdls[0].bl94: 0 [BL 94]
num_codes: 3 [BL 75]
codes[0].bl95: 0 [BL 95]
codes[0].bs76: 128 [BS 76]
codes[0].bl94: 499176513 [BL 94]
codes[1].bl95: 1694528000 [BL 95]
codes[1].bs76: 53249 [BS 76]
codes[1].bl94: 2147889158 [BL 94]
codes[2].bl95: 105944080 [BL 95]
codes[2].bs76: 256 [BS 76]
codes[2].bl94: 0 [BL 94]
action_offset_x: -2.53417e-212 [BD 140]
action_offset_y: 2.2612e-130 [BD 141]
angle_offset: 1.05211e-286 [BD 0]
 handle stream: +92 @124.1 MISSING (@135.6  @149.6)
 padding: +2
 object_map{4B0} = 140
 padding: 80/0 (6 bits)
crc: 858E [RSx]
 check_CRC 15739-15892 = 153: 858E == 858E
```

michal-josef-spacek, 2024-12-02:
I analyzed it and it seems that the handles list continues with ```BL 94``` after handles in the situation when a count of handles is 1.

michal-josef-spacek, 2024-12-02:
In DXF the sequence is:
```
...
 74
2
 94
0
 94
3
 75
0
...
```
The mistake in the previous DWG output is that ```DXF 94``` isn't here, and ```DXF 75``` = 3, and looking for three codes that are not present.

michal-josef-spacek, 2024-12-02:
This was probably a mistake for the code before https://github.com/LibreDWG/libredwg/issues/1049, where the handle contains DXF 74, 94, and 94.

michal-josef-spacek, 2024-12-02:
Hm,  num_handles == 1 condition is false.

michal-josef-spacek, 2024-12-02:
Example #1
```
--common_size: 29
unknown_bits [1161 (29,1057,430) 146 TF]: 3FFFFFFFD1D41CF1D84251D41E25A7632EB9D0D9100AA483C5AEC2985C0903405014090240BD31232F40BCA110319634E4381E21100A2393B0B94A5F

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 7 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 9 [BL 0]
name: "Stretch" [TU 300]
be_major: 29 [BL 98]
be_minor: 7 [BL 99]
eed1071: 0 [BL 1071]
display_location: (162.122, -677.24, 0) [3BD 0]
num_deps: 3 [BL 71]
deps[0]: (4.2.4AD) abs:1197 [H* 330] => BLOCKLINEARGRIP End Grip
deps[1]: (4.2.4AB) abs:1195 [H* 330] => BLOCKBASEPOINTPARAMETER Base Point
deps[2]: (4.2.4A5) abs:1189 [H* 330] => LWPOLYLINE
num_actions: 1 [BL 70]
actions[0]: 1 [BL 91]
conn_pts[0].code: 2 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 2 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (670.251, 655.11) [2RD 1011]
pts[vcount]: (498.619, -144.805) [2RD 1011]
num_hdls: 1 [BL 73]
hdls[rcount1].hdl: (4.2.4A5) abs:1189 [H 331]
hdls[0].bs74: 2 [BS 74]
hdls[0].bl94: 0 [BL 94]
```
**Next item is BL 94**

michal-josef-spacek, 2024-12-02:
Example 2:
```
--common_size: 29
unknown_bits [1393 (29,1119,430) 175 TF]: 3FFFFFFFD1D41CF1D84211D41E101C2214AA2B05F00E05BA7AE08AB3E4090690140502006968D2EACE234000E200A9853B3B40208EB95DD40720C030

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 7 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 8 [BL 0]
name: "Stretch" [TU 300]
be_major: 29 [BL 98]
be_minor: 7 [BL 99]
eed1071: 0 [BL 1071]
display_location: (-5.91861, 30.6681, 0) [3BD 0]
num_deps: 6 [BL 71]
deps[0]: (4.2.C9F) abs:3231 [H* 330] => BLOCKLINEARGRIP End Grip
deps[1]: (4.2.C88) abs:3208 [H* 330] => LINE
deps[2]: (4.2.C87) abs:3207 [H* 330] => LINE
deps[3]: (4.2.C86) abs:3206 [H* 330] => LINE
deps[4]: (4.2.C85) abs:3205 [H* 330] => LINE
deps[5]: (4.2.C84) abs:3204 [H* 330] => LWPOLYLINE
num_actions: 0 [BL 70]
conn_pts[0].code: 1 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 1 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (9.90414, 27.2325) [2RD 1011]
pts[vcount]: (-8.01529, 19.2503) [2RD 1011]
num_hdls: 5 [BL 73]
hdls[rcount1].hdl: (4.2.C84) abs:3204 [H 331]
hdls[0].bs74: 2 [BS 74]
hdls[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.C85) abs:3205 [H 331]
hdls[1].bs74: 3 [BS 74]
hdls[1].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.C86) abs:3206 [H 331]
hdls[2].bs74: 1 [BS 74]
hdls[2].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.C87) abs:3207 [H 331]
hdls[3].bs74: 1 [BS 74]
hdls[3].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.C88) abs:3208 [H 331]
hdls[4].bs74: 1 [BS 74]
hdls[4].bl94: 1 [BL 94]
```
**Next item is BL 94**

michal-josef-spacek, 2024-12-02:
Example 3:
```
--common_size: 21
unknown_bits [1169 (21,1033,350) 147 TF]: 3FFFFFFFD1D504F1D84DD1D5062DFD69B6F08212500725B6FBA2D8C00C09049334CD029847A442348C4D40A0A4BB8C21E935C0B060C7730DA5394000

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 65 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 55 [BL 0]
name: "Stretch6" [TU 300]
be_major: 29 [BL 98]
be_minor: 65 [BL 99]
eed1071: 0 [BL 1071]
display_location: (50.0684, -2.06845, 0) [3BD 0]
num_deps: 4 [BL 71]
deps[0]: (4.1.4E) abs:78 [H* 330] => LINE
deps[1]: (4.1.4A) abs:74 [H* 330] => LINE
deps[2]: (4.1.4D) abs:77 [H* 330] => LINE
deps[3]: (4.1.62) abs:98 [H* 330] => LWPOLYLINE
num_actions: 0 [BL 70]
conn_pts[0].code: 51 [BL 92]
conn_pts[0].name: "XDelta" [TU 301]
conn_pts[1].code: 51 [BL 93]
conn_pts[1].name: "YDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (59.0953, -21.9107) [2RD 1011]
pts[vcount]: (25.6447, 21.7333) [2RD 1011]
num_hdls: 4 [BL 73]
hdls[rcount1].hdl: (4.1.4A) abs:74 [H 331]
hdls[0].bs74: 2 [BS 74]
hdls[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.4D) abs:77 [H 331]
hdls[1].bs74: 1 [BS 74]
hdls[1].bl94: 2 [BL 94]
hdls[rcount1].hdl: (4.1.4E) abs:78 [H 331]
hdls[2].bs74: 0 [BS 74]
hdls[2].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.1.62) abs:98 [H 331]
hdls[3].bs74: 2 [BS 74]
hdls[3].bl94: 0 [BL 94]
```
**Next item is BL 75**

michal-josef-spacek, 2024-12-02:
Example 4:
```
--common_size: 29
unknown_bits [3161 (29,1541,446) 396 TF]: 3FFFFFFFD21610F1D865121612321921D53F6A30EFC893385453C27E73F92298D635021C3C3D1983EF004004CFB5A653E6EE3FC0A6F6A03E09B2BFAC

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 33 [BL 98]
evalexpr.minor: 132 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 148 [BL 0]
name: "Stretch1" [TU 300]
be_major: 33 [BL 98]
be_minor: 132 [BL 99]
eed1071: 0 [BL 1071]
display_location: (-0.153595, 0.723539, 0) [3BD 0]
num_deps: 34 [BL 71]
deps[0]: (4.2.233) abs:563 [H* 330] => BLOCKLINEARGRIP End Grip
deps[1]: (4.2.2C6) abs:710 [H* 330]
deps[2]: (4.2.2C2) abs:706 [H* 330]
deps[3]: (4.2.2C0) abs:704 [H* 330]
deps[4]: (4.2.2BC) abs:700 [H* 330]
deps[5]: (4.2.2B0) abs:688 [H* 330]
deps[6]: (4.2.2AC) abs:684 [H* 330]
deps[7]: (4.2.2A6) abs:678 [H* 330]
deps[8]: (4.2.2A2) abs:674 [H* 330]
deps[9]: (4.2.29F) abs:671 [H* 330]
deps[10]: (4.2.29D) abs:669 [H* 330]
deps[11]: (4.2.299) abs:665 [H* 330]
deps[12]: (4.2.297) abs:663 [H* 330]
deps[13]: (4.2.293) abs:659 [H* 330]
deps[14]: (4.2.28D) abs:653 [H* 330]
deps[15]: (4.2.289) abs:649 [H* 330]
deps[16]: (4.2.26E) abs:622 [H* 330]
deps[17]: (4.2.26F) abs:623 [H* 330]
deps[18]: (4.2.264) abs:612 [H* 330]
deps[19]: (4.2.277) abs:631 [H* 330]
deps[20]: (4.2.273) abs:627 [H* 330]
deps[21]: (4.2.278) abs:632 [H* 330]
deps[22]: (4.2.27C) abs:636 [H* 330]
deps[23]: (4.2.280) abs:640 [H* 330]
deps[24]: (4.2.292) abs:658 [H* 330]
deps[25]: (4.2.28E) abs:654 [H* 330]
deps[26]: (4.2.26A) abs:618 [H* 330]
deps[27]: (4.2.25F) abs:607 [H* 330]
deps[28]: (4.2.25B) abs:603 [H* 330]
deps[29]: (4.2.260) abs:608 [H* 330]
deps[30]: (4.2.288) abs:648 [H* 330]
deps[31]: (4.2.284) abs:644 [H* 330]
deps[32]: (4.2.269) abs:617 [H* 330]
deps[33]: (4.2.265) abs:613 [H* 330]
num_actions: 0 [BL 70]
conn_pts[0].code: 141 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 141 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (2.11695, 0.965616) [2RD 1011]
pts[vcount]: (-0.0704536, 0.674702) [2RD 1011]
num_hdls: 33 [BL 73]
hdls[rcount1].hdl: (4.2.2C0) abs:704 [H 331]
hdls[0].bs74: 1 [BS 74]
hdls[0].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.2C2) abs:706 [H 331]
hdls[1].bs74: 1 [BS 74]
hdls[1].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.2C6) abs:710 [H 331]
hdls[2].bs74: 1 [BS 74]
hdls[2].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.25B) abs:603 [H 331]
hdls[3].bs74: 1 [BS 74]
hdls[3].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.25F) abs:607 [H 331]
hdls[4].bs74: 1 [BS 74]
hdls[4].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.260) abs:608 [H 331]
hdls[5].bs74: 1 [BS 74]
hdls[5].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.264) abs:612 [H 331]
hdls[6].bs74: 1 [BS 74]
hdls[6].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.265) abs:613 [H 331]
hdls[7].bs74: 1 [BS 74]
hdls[7].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.269) abs:617 [H 331]
hdls[8].bs74: 1 [BS 74]
hdls[8].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.26A) abs:618 [H 331]
hdls[9].bs74: 1 [BS 74]
hdls[9].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.26E) abs:622 [H 331]
hdls[10].bs74: 1 [BS 74]
hdls[10].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.26F) abs:623 [H 331]
hdls[11].bs74: 1 [BS 74]
hdls[11].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.273) abs:627 [H 331]
hdls[12].bs74: 1 [BS 74]
hdls[12].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.277) abs:631 [H 331]
hdls[13].bs74: 1 [BS 74]
hdls[13].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.278) abs:632 [H 331]
hdls[14].bs74: 1 [BS 74]
hdls[14].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.27C) abs:636 [H 331]
hdls[15].bs74: 1 [BS 74]
hdls[15].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.280) abs:640 [H 331]
hdls[16].bs74: 1 [BS 74]
hdls[16].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.284) abs:644 [H 331]
hdls[17].bs74: 1 [BS 74]
hdls[17].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.288) abs:648 [H 331]
hdls[18].bs74: 1 [BS 74]
hdls[18].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.289) abs:649 [H 331]
hdls[19].bs74: 1 [BS 74]
hdls[19].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.28D) abs:653 [H 331]
hdls[20].bs74: 1 [BS 74]
hdls[20].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.28E) abs:654 [H 331]
hdls[21].bs74: 1 [BS 74]
hdls[21].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.292) abs:658 [H 331]
hdls[22].bs74: 1 [BS 74]
hdls[22].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.293) abs:659 [H 331]
hdls[23].bs74: 1 [BS 74]
hdls[23].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.297) abs:663 [H 331]
hdls[24].bs74: 1 [BS 74]
hdls[24].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.299) abs:665 [H 331]
hdls[25].bs74: 1 [BS 74]
hdls[25].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.29D) abs:669 [H 331]
hdls[26].bs74: 1 [BS 74]
hdls[26].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.29F) abs:671 [H 331]
hdls[27].bs74: 1 [BS 74]
hdls[27].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.2A2) abs:674 [H 331]
hdls[28].bs74: 1 [BS 74]
hdls[28].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.2A6) abs:678 [H 331]
hdls[29].bs74: 1 [BS 74]
hdls[29].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.2AC) abs:684 [H 331]
hdls[30].bs74: 1 [BS 74]
hdls[30].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.2B0) abs:688 [H 331]
hdls[31].bs74: 1 [BS 74]
hdls[31].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.2BC) abs:700 [H 331]
hdls[32].bs74: 1 [BS 74]
hdls[32].bl94: 0 [BL 94]
```
**Next item is BL 75**

michal-josef-spacek, 2024-12-02:
Example 5:
```
--common_size: 29
unknown_bits [1241 (29,1037,414) 156 TF]: 3FFFFFFFD1D418F1D84491D41A21FD358CB60EABB00F47E627315A543C090590B42D022083E0B4E674AFC042680B45FF21754078A6E6215838B1C07D

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 6 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 18 [BL 0]
name: "Etirer" [TU 300]
be_major: 29 [BL 98]
be_minor: 6 [BL 99]
eed1071: 0 [BL 1071]
display_location: (-3869.42, -39.2897, 0) [3BD 0]
num_deps: 5 [BL 71]
deps[0]: (4.2.58E) abs:1422 [H* 330] => BLOCKLINEARGRIP Poignée d'extrémité
deps[1]: (4.2.5A4) abs:1444 [H* 330]
deps[2]: (4.2.589) abs:1417 [H* 330] => BLOCKROTATIONGRIP Poignée
deps[3]: (4.2.5A3) abs:1443 [H* 330]
deps[4]: (4.2.5A2) abs:1442 [H* 330]
num_actions: 0 [BL 70]
conn_pts[0].code: 11 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 11 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (-4026.45, 338.125) [2RD 1011]
pts[vcount]: (-4408.34, -243.693) [2RD 1011]
num_hdls: 3 [BL 73]
hdls[rcount1].hdl: (4.2.5A2) abs:1442 [H 331]
hdls[0].bs74: 1 [BS 74]
hdls[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.5A3) abs:1443 [H 331]
hdls[1].bs74: 1 [BS 74]
hdls[1].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.2.5A4) abs:1444 [H 331]
hdls[2].bs74: 1 [BS 74]
hdls[2].bl94: 0 [BL 94]
```
**Next item is BL 75**

michal-josef-spacek, 2024-12-02:
Example 6:
```
--common_size: 29
unknown_bits [1185 (29,1077,430) 149 TF]: 3FFFFFFFD1D494F1D84211D49617C42BD2D490FC4FC5FD68F688B53E9BF90290140502601A587908E8F43F00E4D97D318CCDBF0064C479844CEC3FC0

evalexpr.parentid: -1 [BLd 0]
evalexpr.major: 29 [BL 98]
evalexpr.minor: 37 [BL 99]
evalexpr.value_code: -9999 [BSd 70]
evalexpr.nodeid: 8 [BL 0]
name: "Stretch" [TU 300]
be_major: 29 [BL 98]
be_minor: 37 [BL 99]
eed1071: 0 [BL 1071]
display_location: (1.07894, -0.791448, 0) [3BD 0]
num_deps: 2 [BL 71]
deps[0]: (4.2.D1B8) abs:53688 [H* 330] => LWPOLYLINE
deps[1]: (4.2.D1BA) abs:53690 [H* 330] => WIPEOUT
num_actions: 0 [BL 70]
conn_pts[0].code: 1 [BL 92]
conn_pts[0].name: "EndXDelta" [TU 301]
conn_pts[1].code: 1 [BL 93]
conn_pts[1].name: "EndYDelta" [TU 302]
num_pts: 2 [BL 72]
pts[vcount]: (1.30665, -0.230841) [2RD 1011]
pts[vcount]: (0.884341, 1.54313) [2RD 1011]
num_hdls: 2 [BL 73]
hdls[rcount1].hdl: (4.2.D1B8) abs:53688 [H 331]
hdls[0].bs74: 2 [BS 74]
hdls[0].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.2.D1BA) abs:53690 [H 331]
hdls[1].bs74: 2 [BS 74]
hdls[1].bl94: 2 [BL 94]
```
**Next item is BL 75**

michal-josef-spacek, 2024-12-02:
Hm, it seems, that I was wrong with https://github.com/LibreDWG/libredwg/issues/1049.
There is the possibility of multiple ```DXF 94``` in each handle.

michal-josef-spacek, 2024-12-02:
Fine.. 
DXF 74 is count for DXF 94 in handle.
Example output:
```
num_hdls: 11 [BL 73]
hdls[rcount1].hdl: (4.1.69) abs:105 [H 331]
hdls[0].bs74: 1 [BS 74]
hdls[0].bl94s[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.6E) abs:110 [H 331]
hdls[1].bs74: 2 [BS 74]
hdls[1].bl94s[0].bl94: 0 [BL 94]
hdls[1].bl94s[1].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.1.6F) abs:111 [H 331]
hdls[2].bs74: 4 [BS 74]
hdls[2].bl94s[0].bl94: 1 [BL 94]
hdls[2].bl94s[1].bl94: 2 [BL 94]
hdls[2].bl94s[2].bl94: 3 [BL 94]
hdls[2].bl94s[3].bl94: 4 [BL 94]
hdls[rcount1].hdl: (4.1.70) abs:112 [H 331]
hdls[3].bs74: 2 [BS 74]
hdls[3].bl94s[0].bl94: 0 [BL 94]
hdls[3].bl94s[1].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.1.72) abs:114 [H 331]
hdls[4].bs74: 2 [BS 74]
hdls[4].bl94s[0].bl94: 0 [BL 94]
hdls[4].bl94s[1].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.1.76) abs:118 [H 331]
hdls[5].bs74: 1 [BS 74]
hdls[5].bl94s[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.77) abs:119 [H 331]
hdls[6].bs74: 2 [BS 74]
hdls[6].bl94s[0].bl94: 0 [BL 94]
hdls[6].bl94s[1].bl94: 1 [BL 94]
hdls[rcount1].hdl: (4.1.7A) abs:122 [H 331]
hdls[7].bs74: 1 [BS 74]
hdls[7].bl94s[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.7B) abs:123 [H 331]
hdls[8].bs74: 1 [BS 74]
hdls[8].bl94s[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.7E) abs:126 [H 331]
hdls[9].bs74: 1 [BS 74]
hdls[9].bl94s[0].bl94: 0 [BL 94]
hdls[rcount1].hdl: (4.1.7F) abs:127 [H 331]
hdls[10].bs74: 1 [BS 74]
hdls[10].bl94s[0].bl94: 0 [BL 94]
```

@rurban What do you think about naming?
We could rename bs74 to something num_, sure?

rurban, 2024-12-03:
Let's rename bs74 to num_index, bl94 to index, and bl94s to indices.

michal-josef-spacek, 2024-12-03:
@rurban ok, I will prepare PR. 

The same situation will be for the codes section:
```
num_codes: 1 [BL 75]
codes[0].bl95: 8 [BL 95]
codes[0].bs76: 2 [BS 76]
codes[0].bl94: 0 [BL 94]
```

michal-josef-spacek, 2024-12-03:
The actual output (#1054) of codes:
```
...
num_codes: 4 [BL 75]
codes[0].bl95: 17 [BL 95]
codes[0].num_indexes: 1 [BS 76]
codes[0].indexes[0].index: 0 [BL 94]
codes[1].bl95: 61 [BL 95]
codes[1].num_indexes: 2 [BS 76]
codes[1].indexes[0].index: 0 [BL 94]
codes[1].indexes[1].index: 1 [BL 94]
codes[2].bl95: 69 [BL 95]
codes[2].num_indexes: 2 [BS 76]
codes[2].indexes[0].index: 0 [BL 94]
codes[2].indexes[1].index: 1 [BL 94]
codes[3].bl95: 75 [BL 95]
codes[3].num_indexes: 1 [BS 76]
codes[3].indexes[0].index: 1 [BL 94]
...
```

michal-josef-spacek, 2024-12-04:
@rurban PR is fine from my side :-)

michal-josef-spacek, 2024-12-07:
WIP PR: https://github.com/LibreDWG/libredwg/pull/1055

michal-josef-spacek, 2024-12-08:
@rurban 
I tested it and the situation with logging is the same as in my PR:
```
...
num_codes: 4 [BL 75]
codes[0].bl95: 17 [BL 95]
codes[0].num_indexes: 1 [BS 76]
indexes: { 0 } [*BL 94]
codes[1].bl95: 61 [BL 95]
codes[1].num_indexes: 2 [BS 76]
indexes: { 0 1 } [*BL 94]
codes[2].bl95: 69 [BL 95]
codes[2].num_indexes: 2 [BS 76]
indexes: { 0 1 } [*BL 94]
codes[3].bl95: 75 [BL 95]
codes[3].num_indexes: 1 [BS 76]
indexes: { 1 } [*BL 94]
...
```

Missing ```codes[3].``` in indexes item.