LibreDWG/libredwg #1062: Issue in ACAD_TABLE entity in AC1015

michal-josef-spacek, 2024-12-23:
In https://github.com/LibreDWG/libredwg/issues/1059 example [table.zip](https://github.com/user-attachments/files/18229987/table.zip) is issue:
```
--common_size: 14490
unknown_bits [3172 (14490,2891,0) 397 TF]: 0000000000101E5000000000000004340BA922CD20C820000000000000508000000000000014200000000000000608000000000000014200000000000000508000000000000018200000
00000000038800000000000000E20000000000000038800000000000000E202020020C80C832A24AA2622805210040400000000000000404040240501AA020082A020120280D501004150100901406A80802028080480A03540401054040240
501AA020082A020020280D501004050100101406A808020A8080080A03540401054040040501AA020080A020020280D501004150100101406A808020A8080080A0354040101404004050190A736F6D65207465787400A420080800000000000
1F07E8080080A03540401014040040501AA020080A020020280D501004050100101406A80802028080080A03540401014040040501AA020082A020020280D501004150100101406A80802028080080A03540401054040040501AA0200828191
04B20202888290145A90F0C2828282828282828282828282828282828282828282828280E

ins_pt: (404, 38, 0) [3BD 10]
scale_flag: 3 [BB 0]
scale: (1, 1, 1) [DD 41]
rotation: 0 [BD 50] 0º
extrusion: (0, 0, 1) [3BD 210]
has_attribs: 0 [B 66]
block_header: (5.2.28B) abs:651 [H 2] => BLOCK_HEADER *T
tablestyle: (5.2.1E18) abs:7704 [H 342]
flag_for_table_value: 22 [BS 90]
horiz_direction: (1, 0, 0) [3BD 11]
num_cols: 6 [BL 92]
num_rows: 4 [BL 91]
col_widths[0]: 12 [BD 142]
col_widths[1]: 12 [BD 142]
col_widths[2]: 16 [BD 142]
col_widths[3]: 12 [BD 142]
col_widths[4]: 12 [BD 142]
col_widths[5]: 16 [BD 142]
row_heights[0]: 7 [BD 141]
row_heights[1]: 7 [BD 141]
row_heights[2]: 7 [BD 141]
row_heights[3]: 7 [BD 141]
ERROR: Invalid ACAD_TABLE.cells x 24

 object_map{E11} = 1209
 padding: 2/2 (3 bits)
crc: 87A4 [RSx]
 check_CRC 76046-78258 = 2212: 87A4 == 87A4
```

rurban, 2024-12-23:
Of course, it's unstable. Thanks for the test case, but we already do have plenty of such TABLE`s already

michal-josef-spacek, 2025-01-03:
I am opening this ticket for fixing the TABLE entity in AC1015.

michal-josef-spacek, 2025-01-15:
@rurban Do you have any hint to the previous question?

michal-josef-spacek, 2025-03-09:
I know how to do (to previous question).

rurban, 2025-03-10:
Use both. 2 is spezialized to lookup the name

michal-josef-spacek, 2025-03-11:
@rurban Yes, I found this. I am working on this step by step.

rurban, 2026-11-08:
Fixed, with the `table.dwg` of #1059 (AC1015, 6 columns x 4 rows).

**The decode error** (`Invalid ACAD_TABLE.cells x 24`, the rest of the entity left as unknown bits): the cell alignment, present with bit 0x01 of the cell flag override, is a bitshort (BS), not a raw short (RS) as the ODA spec says. Everything after the first cell with an alignment override was garbage. Fixed on master in da34cdf6a ("TABLE: the pre-R2007 cell alignment is a BS, not a RS"). All 24 cells now decode with the same texts, flags, merges, text heights (2.0, 1.5) and alignments as ODAFileConverter reads them (it only omits the default alignment 1), without errors, and a `dwgrewrite` roundtrip reads back the same.

**DXF 2 and 343:** as answered above, write both. `2` is the block name (`FIELD_HANDLE (block_header, 5, 2)` writes the name), and the handle of the anonymous `*T` block is written as `343` in the AcDbTable subclass, after `342`. Our DXF for this drawing now has `2 *T115` and `343 28B`, as ODA writes it. That is 3e45ca6d9 ("dxf: write the TABLE block record handle as 343"), which is on its way to master after CI. dxf2dwg reads it back with the same block, found by the name.

ODA also writes the table flags 93-96 and a different 90 flag value, which we do not yet. They are not needed to read the table, so they are not part of this issue. Please open a new issue if they matter for you. Thanks for the test case and for the step by step work on this.

closed 2026-11-08 by rurban