LibreDWG/libredwg #1065: Autocad can't open dwg writen by libredwg

lhbose, 2024-12-25:
#The question is that i can't open dwg file writen by libredwg with any version of autocad,can anyone help me?
#Here is my mini-testing code:

#include <stdio.h>
#include "dwg.h"
#include "dwg_api.h"

int main(int argc,char* argv[])
{
    int error;
    const char* szFilePath = "/root/libredwg_test/test.dwg";
    
    Dwg_Data* pDwgData = dwg_new_Document(R_2000,0,2);
    
    error = dwg_write_file(szFilePath, pDwgData);
    
    dwg_free(pDwgData);

    return 0;
}

michal-josef-spacek, 2024-12-30:
I could probably test and fix this issue.
The issue is because some crucial default bad value in header/aux_header/?.

I think the same issue as https://github.com/LibreDWG/libredwg/issues/1058

rurban, 2025-01-20:
Not repro. DWG and DXF files written by LibreDWG are regularily tested with ODA in the CI. Esp. simple ones with R_2000.

You might encounter a RECOVER warning though.

michal-josef-spacek, 2025-01-20:
This is the bug. There is no same situation in the header and aux header. The question is where. In multiple places definitely.

michal-josef-spacek, 2025-01-21:
I checked the header, thumbnail, aux_header, second_header and there isn't an issue with the recovery probably. There are some issues, but not this one.

So the question is where is the issue? Or how to fix? :-/

michal-josef-spacek, 2025-01-22:
The issue begin with R_13 version.

rurban, 2025-02-26:
I'm checking the dwgadd api now:

for dwgadd.example_r11 these api's create wrong objects:
- [ ] polyline_2d
- [ ] polyline_pface
- [ ] dimstyle
- [ ] ucs

rurban, 2025-03-04:
The example works fine, there are no dwg_write errors, and no ODA conversion errors.

api.c:
``` c
#include "dwg.h"
#include "dwg_api.h"

int main(int argc,char* argv[])
{
    int error;
    const char* szFilePath = "test_2000.dwg";
    Dwg_Data* pDwgData = dwg_new_Document(R_2000, 0, 3);
    error = dwg_write_file(szFilePath, pDwgData);
    dwg_free(pDwgData);
    return error;
}
```
``` sh
cc -I../include api.c src/.libs/libredwg.a -lm
./a.out
ODAFileConverter . . ACAD2000 DXF 0 1 test_2000.dwg
cat test_2000.dxf.err
cat: test_2000.dxf.err: No such file or directory
```

closed 2025-03-04 by rurban

michal-josef-spacek, 2025-03-24:
@timoria21 Yes this issue is here. The issue is very complex. Mainly related to the bad setting of defaults in header variables in LibreDWG. Or/and more.

e.g. I created the ticket to fix AC1009 DXF which is related to it. (https://github.com/LibreDWG/libredwg/issues/1100).