LibreDWG/libredwg #1071: Add an 3D Export

Mannshoch, 2025-01-07:
In my case, I mostly have to work with 3D-DWG files.

I would like to have a possibility to extract 3D Data.  The file format is not that important ( could be either, sat, step, iges, ifc )

Objects to extract: Surfaces, 3D-Body, meshes, etc.

Structure  should be in any case an Assembly export

(e.g. STEP):
* FileName
  * Layer-Name 1
     * Layer-Name 1- BodyName1
     * Layer-Name 1 - BodyName2
     * Layer-Name 1 - SurfaceName1
     * Layer-Name 1 - 3D-shapeName
  * Layer-Name 2
     *  XREF1-Filename
        * Layer-Name D
           * Layer-Name D - BodyName1
        * Layer-Name A
          * Layer-Name A - BodyName2
        * Layer-Name A
          * Layer-Name A - SurfaceName1

rurban, 2025-01-07:
That's too complicated and out of scope. We're export DWG, DXF or some private JSON.
Other packages can go from there, eg assimp

rurban, 2025-01-13:
That's interesting. 
Have you compared the AutoCAD SAT export with the SAT bodies from libredwg, e.g. in JSON? Because they can be in a special AcDsPrototype_1b section since 2013, as SAB binary ACIS data, version 2.
Which we also decode properly.

Mannshoch, 2025-01-13:
Sorry but I'm a CAD user not a developer. Your mention of JSON was interesting but I don't know what to do here.

I attached you an example. I imported the sat-file into a dwg.
[Waschbecken.zip](https://github.com/user-attachments/files/18397978/Waschbecken.zip)

Mannshoch, 2025-01-15:
what DXF version do you wish?

What kind of problems are there? As far that kind of data I usually have to work with. It's ok if they have error as long I could get them soemhow. I may could repair it with FreeCAD or other applications.

Mannshoch, 2025-01-16:
[Waschbecken2.zip](https://github.com/user-attachments/files/18440454/Waschbecken2.zip)

- Waschbecken-orig.sat was the Original I used
- Waschbecken-export.sat is the export I did after I imported it into dwg.
- Waschbecken-mdt2025.dwg is the original dwg My AutoCAD 2025 Mechanical Desktop creates.
- Waschbecken-2018.dwg is format 2018 from the save as dialog
- Waschbecken-2018.dxf is the 2018 DXF format.

michal-josef-spacek, 2025-01-16:
@Mannshoch Thank you very much for testing data.

We have actually issues with:
* 3DSOLID entity, probably resolvable, in entity data is probably link to database.
* Decoding of "AcDs datastorage", not simple thing, SAT file is probably inside it. In DXF are some hints.
* There are some XRECORD objects, I don't know what they are.

If you want to help us, you could create some small sat file with simple 3d object and insert it to autocad and send to us. (SAT, DWG and DXF files). Working on this big file isn't simple. This issue is not simple still.

Mannshoch, 2025-01-16:
May that is a binary SAT. From SAT files there are not only text files (.sat) also a binary filetype (.sab) does exist. 
https://en.wikipedia.org/wiki/ACIS#File_format
googling for that gave me this here: http://www-isl.ece.arizona.edu/ACIS-docs/PDF/GET/03OV.PDF

michal-josef-spacek, 2025-01-16:
@Mannshoch I understand. There are situations like ACIS in REGION in ASCII form, converted to the next version of DWG and present in BINARY form.

rurban, 2025-01-24:
I can think of exporting decrypted SAT (ACIS) data somehow, but it already easy todo with export to json and filtering the SAT data with jq.

Mannshoch, 2025-01-27:
Thanks for the feedback.
I think from this point on it's more a question about how you want proceed here. 

My Feature request was to have a way to get all 3D-Bodies from dwg Files. 
I'm normal user, for me json is a strange text file. I would use something like dwg2dxf. But it's up do you if you create a dwg2sat or let it be on the json or DXF way.
Most important is, that it is possible to get all 3D-Data and at least the exported DXF should be more complete. Currently I'm working with https://github.com/fougue/mayo to view 3D Data. At moment I have to convert my dwg to dxf for get it work.

rurban, 2025-01-27:
use:
`jq '.OBJECTS[] | select(.acis_data != null) | .acis_data' example_2000.json`
or 
`dwgfilter '.OBJECTS[] | select(.acis_data != null) | .acis_data' example_2000.dwg`

this is essentially your dwg2sat feature

closed 2025-01-27 by rurban