LibreDWG/libredwg #1091: DXF roundtrip issues r2000

rurban, 2025-03-05:
check with dxf-check

open issues:

- [ ] AcDbViewportTable on out_dxf
  OdError thrown during readFile of drawing "blank_2000.dwg" :
    Object improperly read: <AcDbViewportTable> (8)
  Previous error: Object improperly read
  the dxf is written as 2 VPORT 70 1, but should be written as 70 2

rurban, 2025-03-05:
after in_dxf, the num_entries are wrong: 
../to/r13/from_ezdxf/bin_dxf_r13.dwg

Encode object LTYPE_CONTROL
handle: 0.1.5 [H 5]
num_eed: 0
bitsize: 0 [RL 0]
num_reactors: 0 [BL 0]
num_entries: 2 [BS 70]
-bitsize calc from HANDLE_STREAM 72 @9.0 (1588)
ownerhandle: (4.0.0) abs:0 [H 330]
xdicobjhandle: (3.0.0) abs:0 [H 360]
entries[vcount][0]: (2.1.16) abs:22 [H* 0]
entries[vcount][1]: (2.1.16) abs:22 [H* 0]
byblock: (3.1.14) abs:20 [H 0]
bylayer: (3.1.15) abs:21 [H 0]

shoud be 1. same for LAYER_CONTROL, VPORT, APPID (6 instead of 3)

timoria21, 2025-09-04:
Hello, 

Do you happen to have any news on this?

Thank you!

timoria21, 2025-10-06:
Can you please bring this issue forward?

timoria21, 2025-11-05:
Are you back to work? I'm blocked by this issue.

timoria21, 2025-12-16:
Hello Reini, any update on this?

timoria21, 2026-01-14:
Hello there. Happy New Year!

Is there any progress on this topic?

rurban, 2026-02-26:
Not yet. Still too busy with other stuff