LibreDWG/libredwg #1092: JSON reading performance improvement

fr-an-k, 2025-03-06:
I fixed jsmn;, but the project seems dead so if nothing happens after a while, jsmn should probably be forked or copied into LibreDWG with the fix:
https://github.com/zserge/jsmn/pull/241

Parsing my 140MB JSON (generated from a 17 MB DWG) went from 10-20 minutes to a fraction of a second, which is the norm.

rurban, 2025-03-08:
Great. Let's give Serge a week. Then we'll import your code, without submodule

pt300, 2025-03-19:
You are better off forking the code. Zserge abandoned JSMN and I don't really maintain it either. Sorry guys. Maybe one day it will get picked up. Today is not the day.

rurban, 2025-03-22:
I couldn't measure any benefits with the biggest DWG's I had with the first https://github.com/zserge/jsmn/pull/241 patch (fixed in my fork).
Will try to measure  JSMN_PARENT_LINKS next.

fr-an-k, 2025-03-22:
I had the issue for a 130MB JSON, from a typical infrastructure DWG; parsing the tokens became slower and slower as it advanced.