LibreDWG/libredwg #1093: "Recovery Mode" again

timoria21, 2025-03-07:
As you remember, you were unable to resolve the "Recovery Mode" issue when opening DWG files created by LibredDWG in Autodesk AutoCAD. More in detail, currently, it is possible to avoid "Recovery Mode" only with the Acad2000 file format.

The main problem was the specific file format header and material definitions for higher DWG file formats. As we talked about in this issue: https://github.com/LibreDWG/libredwg/issues/996

The question is: if the library below can write files without "Recovery Mode", could you copy from it and resolve the "Recovery Mode" issue for all LibreDWG DWG file formats?

https://github.com/DomCR/ACadSharp

Thank you. Let me know if you are interested and if I can do some DWG test for you.

P.S.: It seems that they had some problems with "Recovery Mode" too, but I cannot tell if they are resolved and for what specific DWG file format: https://github.com/DomCR/ACadSharp/issues/315

michal-josef-spacek, 2025-03-12:
@timoria21
We know about [ACadSharp](https://github.com/DomCR/ACadSharp) project.
And we open to fix issues. :-)

rurban, 2025-03-12:
The only problem with ACadSharp that is not compilable under Linux or MacOS anymore. Windows Visual Studio only. Working on it, but it's a pain

timoria21, 2025-03-13:
My idea was simply to understand if there is something that helps with "unknown" header sections or material definitions that prevents LibreDWG to write "Recovery Mode" free DWG files in all DWG file formats.

Thank you.

michal-josef-spacek, 2025-03-14:
@timoria21 The best help is to create any precisely defined DWG file from AutoCAD with DXF export, which shows some error. And describe that error. 

So in the topic "cannot be opened in AutoCAD" it is possible to create some such examples. e.g. by dwgadd or by LibreDWG library C code or convert DXF file from AutoCAD by dwgwrite with bad DWG output.

michal-josef-spacek, 2025-03-20:
@timoria21 You are talking about some "Recovery Mode" issue. But nothing concrete.

I'm talking about the best way to create a real issue.

rurban, 2025-03-22:
Duplicate of #994 and #996

closed 2025-03-22 by rurban