LibreDWG/libredwg #1100: Bad generation of AC1009 DXF from dwgadd

michal-josef-spacek, 2025-03-22:
I tested AC1009 DXF output of dwgadd for simple line:
```
line (0 0 0) (100 100 100)
```

There are multiple issues:
- [ ] PUCSYDIR with (0,0,0) which is invalid in AutoCAD
- [ ] BLOCK_HEADER table present, this is not possible in AC1009 DXF
- [ ] Missing TEXTSTYLE, CLAYER, CELTYPE and DIMSTYLE values in header
- [ ] Missing MODEL and PAPER space BLOCKS
- [ ] Missing VPORT table

michal-josef-spacek, 2025-03-22:
First two items are fixes in https://github.com/LibreDWG/libredwg/pull/1101

michal-josef-spacek, 2025-04-07:
In case of TEXTSTYLE issue: I created PR (https://github.com/LibreDWG/libredwg/pull/1102), and I don't know how to fix it.

rurban, 2025-06-09:
Tested all DXF and DWG versions from r9-r2000:

dwgadd_line_r9.dxf ok
dwgadd_line_r10.dxf ok
dwgadd_line_r11.dxf ok
dwgadd_line_r13.dxf ok
dwgadd_line_r13.dwg ok
dwgadd_line_r14.dxf ok
dwgadd_line_r14.dwg ok
dwgadd_line_r2000.dxf ok
dwgadd_line_r2000.dwg ok

Broken:
- [ ] Empty Record name with r9-r11 DWG's

closed 2025-06-14 by rurban