LibreDWG/libredwg #1107: hello ,When I use DWG2DXF to parse the DWG file into DXF, the following log information is reported：

leishenglong, 2025-04-02:
log:

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

ERROR: Invalid HATCH.curve_type 0

Warning: Unhandled Object TABLESTYLE in out_dxf 34/7F
Warning: Unstable Class object 503 MATERIAL (0x481) 42/ED
Warning: Unstable Class object 503 MATERIAL (0x481) 43/EE
Warning: Unstable Class object 503 MATERIAL (0x481) 44/EF
Warning: Unstable Class object 508 MLEADERSTYLE (0xfff) 241/90D5
Warning: Unstable Class object 514 DATALINK (0x480) 7204/169BF
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unstable Class object 514 DATALINK (0x480) 7206/169C1
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic
Warning: Unknown object, skipping eed/reactors/xdic

rurban, 2025-04-02:
We'd really need this DWG, please. You can send it in private per email

michal-josef-spacek, 2025-04-05:
It's a good idea to indicate which version of the DWG file it is.

rurban, 2026-05-16:
I can repo it now in the tests