LibreDWG/libredwg #1122: Fix the null reference vulnerability.

QiuYitai, 2025-04-18:
Hello,
Our team has recently been conducting research on a null-pointer-dereference (NPD) vulnerability detection tool and used it to scan libredwg(the version on the master branch). After a manual review, we have identified some potentially vulnerable code snippets that may lead to null-pointer-dereference bugs. 
The NULL Dereference vulnerability happens in `dwg_dynapi_entity_set_value()`, `dynapi.c`
How the NULL Pointer Dereference happens:
1. When `obj == NULL.`
2. NULL dereference of variable `obj` happens at `obj->fixedtype == DWG_TYPE_STYLE`
```
EXPORT bool
dwg_dynapi_entity_set_value (void *restrict _obj, const char *restrict name,
                             const char *restrict fieldname,
                             const void *restrict value, const bool is_utf8)
{
    ......
    const Dwg_Object *obj = dwg_obj_generic_to_object (_obj, &error);
    ......
=>  if (obj && strNE (obj->name, name))//Assuming 'obj' is null
    {
        ......
    }
   {    
       ......
       if (f->is_string && dwg->header.from_version < R_13b1 &&
=>         (obj->fixedtype == DWG_TYPE_STYLE || obj->fixedtype == DWG_TYPE_LTYPE))
       {
           ......
       }
   }
 }
```

rurban, 2025-04-18:
Nice tool, thanks

QiuYitai, 2025-04-25:
Thank you for confirming the NULL pointer dereference vulnerability. To help users track and mitigate this issue, could you please request a CVE ID on [CVE Request Website](https://cveform.mitre.org/).
If you're unable to file this, I can help to submit the request.

rurban, 2025-04-25:
You can do that. But it's only low prio, with an invalid DXF or dwg only. I cannot really think about cases were obj can actually turn up as NULL, and in such cases the process must abort anyway.