LibreDWG/libredwg #1127: Autocad message on open DWG: "need recover" and wrong dwg->header_vars.HANDSEED

MASTER-OF-ORION, 2025-04-25:
Hш guys! I'm saving a DWG file in the R_2000 version, and when I try to open it in AutoCAD, I get a message saying the file is corrupted and requires recovery. I spent a long time digging into this issue, studying the LibreDWG source code and testing in different versions of AutoCAD, and I came to the conclusion that the culprit is this specific parameter:
`dwg->header_vars.HANDSEED``
I also saw the latest commit on this topic:
https://github.com/LibreDWG/libredwg/commit/9ddc300523987a7245db30a3cdd62b662c86ff74
And it seemed like this guy actually fixed it... but nope! Not so fast! Yes, it got better—now an empty drawing or a drawing with four or fewer primitives opens without warnings. But if you draw more than four primitives, say, five lines or five circles—the warning comes back.
I tried manually setting this value to 0xff:
```
dwg->header_vars.HANDSEED = dwg_add_handleref(dwg, 0, UINT64_C(0xff), NULL);
int err = dwg_write_file("c:/0/out2.dwg", dwg);
```
And miracle of miracles—the warning disappeared again! But what's really going on? What exactly does this number control, and how should it be set correctly? If you understand this, I suggest fixing it and releasing another commit, since the previous one didn’t fully solve the problem.

P.S.: Hmm... maybe it's the total number of primitives + some offset? But then one byte is clearly not enough, since a drawing can have up to 100,000 objects.

rurban, 2025-04-25:
The HANDSEED is the handle of the next free object. It should be calculated at save time. But as it look's like this is not working properly yet

rurban, 2025-05-15:
See branch smoke/add-gh1105 and GH #1105