LibreDWG/libredwg #1132: How do I get the text only from the blocks that are inserted into the drawing?

ViX-EL, 2025-05-05:
I need to get the text from the blocks inserted into the drawing, how can I do this?

ViX-EL, 2025-05-05:
Dwg_Data dwg;

memset(&dwg, 0, sizeof(Dwg_Data));
int success = dwg_read_file(filePath.c_str(), &dwg);
if (!(success < DWG_ERR_CRITICAL))
{
	throw std::exception("Wrong file!");
}

int found{ 0 };
for (BITCODE_BL i = 0; i < dwg.num_objects; i++)
{
	Dwg_Object* obj = &dwg.object[i];
	if (obj->type == DWG_TYPE_TEXT || obj->type == DWG_TYPE_MTEXT)
	{
		found++;
		text.append(separator + utf8_decode(process_text(obj)));
	}
}

text = text + separator;

if (!found)
{
	throw std::exception("Text not found!");
}

dwg_free(&dwg);

ViX-EL, 2025-05-05:
Here is my code:

Dwg_Data dwg;

memset(&dwg, 0, sizeof(Dwg_Data));
int success = dwg_read_file(filePath.c_str(), &dwg);
if (!(success < DWG_ERR_CRITICAL))
{
	throw std::exception("Wrong file!");
}

int found{ 0 };
char* text_value{ nullptr };
int isnew{ 0 };
for (BITCODE_BL i = 0; i < dwg.num_objects; i++)
{
	Dwg_Object* obj = &dwg.object[i];
	if (obj->type == DWG_TYPE_TEXT)
	{
		Dwg_Entity_TEXT* textEntity = obj->tio.entity->tio.TEXT;
		if (textEntity && textEntity->text_value) {
			dwg_dynapi_entity_utf8text(textEntity, "TEXT", "text_value", &text_value, &isnew, NULL);
		}
	}
	else if (obj->type == DWG_TYPE_MTEXT)
	{
		Dwg_Entity_MTEXT* mtextEntity = obj->tio.entity->tio.MTEXT;
		if (mtextEntity && mtextEntity->text) {
			dwg_dynapi_entity_utf8text(mtextEntity, "MTEXT", "text", &text_value, &isnew, NULL);
		}
	}
	found++;
	text.append(separator + utf8_decode(text_value));
}

text = text + separator;

if (!found)
{
	throw std::exception("Text not found!");
}

dwg_free(&dwg);

erfansn, 2025-05-17:
You must query the INSERT entity, then iterate on its owned entities